PE5PVB/TEF6686_ESP32 #317: TA indicator always on for EON stations

andimik, 2023-10-15:
I know that the standard says, TP = 0 and TA = 1 means EON, but the TA indicator is always on which is misunderstanding. 

My expectation for such cases is, TA is only shown when another station from EON is currently having an active TA ...

PE5PVB, 2023-10-15:
If you want to have accurate information about the current station it should be like this:

TP - TA
0  0  No traffic service possible at all on or via this service
0  1  Traffic service provided via one or more EON cross-referenced services
1  0  Traffic service provided on this service and also possibly via EON services
1  1  Traffic report in progress on this service

I currently show the status of these flags, so you are correct. Question is, what do you want to show. What I can do is detect in the EON section if any of the stations do have a TA and then set the flag high, or leave it like it is.

lxsxl, 2023-10-15:
I would say keep  the raw indicators as they are because it is conform with the RDS specification. You could probably add further EON information (i. e. TP / TA / PTY) in the EON/AF view. But there is already no space left. Probably divide this screen into two seperate screens? Don't know if it is possible...

PE5PVB, 2023-10-15:
Maybe an seperate EON screen. But this is not an answer on the issue. I'll ask the community about this.

PE5PVB, 2023-10-15:
I'd to ask the community a question regarding the TA symbol.
Currently it shows the TA and TP flag as transmitted by the received radiostation. But when there is EON, the TA flag is always high. I like to know your preference on this.
A. Keep it like it is.
B. Only show TA when a traffic announcement is going on, on the tuned radiostation.
C. Also show TA when a traffic announcement is going on on another radiostation in the EON.

lxsxl, 2023-10-15:
Yes that's right. But RDS specification says that TP = 0 and TA = 1 does only mean that there is an EON service that provides traffic information and not that it actually transmits those information. If this information is also needed you could probably change the color of the TA-indicator if TP = 0 AND TA = 1 AND EON-TP = 1 AND EON-TA = 1.

lxsxl, 2023-10-15:
As the majority tends to B., I have two questions:
1. What is the trigger for the EON-indicator, presence of group 14A or TP=0 and TA=1?
2. What is the trigger for the TA-indicator, TA=1 in group 0A or in group 15B or the first received of those?

PE5PVB, 2023-10-15:
1. When Group 14A is present
2. Bit 4 in block B in group 0A or 0B

mtv-alex, 2023-10-16:
If we talk about a household receiver (car radio), then the indication of this flag goes as TR (when TR = 0, TA = 1). Since I have two stations transmitting TR=0/TA=1, I didn't see any indication on the receiver display.
That's why I pointed to this point. As for me, the flag indication can be left. If traffic information is transmitted at the station, only TR will be lit. And at the time of traffic information - TR and TA. Stations where only the ONE is lit simply informs that there is a switch to another station during traffic information. TR=0/TA=1 does not make any adjustments to the operation of the receiver (for example, increasing the volume) until the command to switch comes.

andimik, 2023-10-16:
TR?

PE5PVB, 2023-10-16:
Almost every station has TP set true, because otherwise a lot of radio's won't stop on that station during channel seeking.

mtv-alex, 2023-10-16:
As for me, there should be some kind of identification. Or as in car radios (when TP=0/TA=1, TR is lit on the display. But then somewhere really show to show that it is the expression TP=0/TA=1 that is accepted) or leave it as it is now.
Ignore this point is not worth it. There are many such stations in Europe broadcasting according to this scheme.
It is incorrect to write TA/EON, because it means that the 15B group is being transmitted. And it works at the moment when we want to switch the listener to a station with traffic information.

PE5PVB, 2023-10-16:
Didn't you mix up TP (Traffic Program) and TA (Traffic announcement)?

Well, in EON information I can get the actual program which is broadcasting the TA.

mtv-alex, 2023-10-16:
Maybe we're starting to get a little confused. The question is: whether to show the burning TA flag when the TP flag is inactive (expression TP=0/TA=1). Do I understand the problem correctly? If this is the case, then I suggest not to change anything - leave it like this now. This situation should also be displayed. At the moment, the receiver is showing as it should.

PE5PVB, 2023-10-16:
No, discussion is about show the flags as transmitted, or show the flags as they should be processed.
If TP is 0 and TA is 1, the radio knows that there could be another station in the EON who will transmit traffic announcements.

When displaying the flags as transmitted, in this case the TA is always on, which could suggest that a traffic annoucement is going on.

PE5PVB, 2023-10-16:
Yes, but the noob user will think that there is a problem because the traffic annoucement flag is still on while music is played.

PE5PVB, 2023-10-17:
For now I will leave it as it is. Just show the raw flags.

closed 2023-10-17 by PE5PVB

mtv-alex, 2023-10-17:
I also want to add: they wrote that the combination TP = 0 / TA = 1 is only strictly shown when there is a 14 A group (EON). Ideally and according to the standard, yes. I attach two photos that it is possible to transmit the combination TP = 0 / TA = 1 having an EON, or simply not having it. I just turned off group 14A at home to show an example. There may also be such cases - engineers have configured something wrong or something else. There are all sorts of situations.
[image]
[image]