PE5PVB/TEF6686_ESP32 #321: Problems with the function: Touch rotating

mx4zxc, 2023-10-17:
I have noticed that the "Touch rotating" function no longer works. 

I don't really know what this function actually is, but I used it often in the AM band. When I set Touch rotating ON, I could tune with the tuning knob even if the display was OFF. 
It was very useful for finding weak stations as noise from the display could wipe these stations out completely.

andimik, 2023-10-17:
Me too. I have never understood that menu option. Therefore, I could not translate into German.

ohmytime, 2023-10-17:
It means :
You can still change frequency when screen is off.

I just add this function back #323

ohmytime, 2023-10-17:
> So let's use another menu text for that.

OK， you can translate to your local language , to make it easier to understand.

mx4zxc, 2023-10-17:
Thanks so much :) It works fine now. I have the gray china model and it produces so much noise from the display that it is no use searching for stations on the LW and MW band with the display on. Even on the SW band, the weak stations disappear with the display switched on.
Andimic:
I have translated it roughly like this: Allow to tune with display off.

andimik, 2023-10-17:
No I mean that even the English text is not well understandable.

closed 2023-10-17 by mx4zxc