PE5PVB/TEF6686_ESP32 #330: Error in the interface when switching saved radio stations

Yurij20, 2023-10-20:
When switching saved radio stations, sometimes the interface in this part of the display fails.  Circled in the photo below
[image]

Yurij20, 2023-10-20:
[image]

andimik, 2023-10-20:
[image]

BTW

You can attach pictures (and short videos and files) also here in github.com. In some browsers you need to switch to desktop view in order to do that...

PE5PVB, 2023-10-23:
Can you get me a way to reproduce this? What frequencies at what preset and how to reproduce?

Yurij20, 2023-10-30:
This problem appeared periodically, so there is no way to reproduce it.  I installed the latest firmware release and it doesn't seem to show up anymore

PE5PVB, 2023-11-01:
Well. Possibly fixed on the way. if it occurs again, please reopen with reproduction plan.

closed 2023-11-01 by PE5PVB