PE5PVB/TEF6686_ESP32 #332: Unsorted AF broken

mrwish7, 2023-10-20:
I'm not sure since when exactly, but in the most recent beta versions, disabling AF sort breaks the AF list - what seems to be shown is broadcast order but not from the first element, then a 0.0 value and the broadcast order. Also, clicking the bottom button to go to page 2/2 (when on the second AF/EON detailed screen) then makes the receiver soft reboot. I'm guessing the recent tweaks for method B are what has caused this change. The AF list in question here is a UK one, using method A.

[image]

Broadcast order from RDS Spy -

[image]

Originally when this feature was added it did work (in this example with 89.9 consistently in position 1), and I also made a further mod for this setting to mean that the current frequency wasn't removed from the list when sorting was disabled - https://github.com/PE5PVB/TEF6686_ESP32/pull/267

Switching AF auto sort to on makes the second AF/EON detailed screen work correctly, with the sorted AF list as expected.

I won't have chance to look into this myself for a couple of days, so I'm just raising it as an issue in case it's an easy fix back (without breaking any of the new method B functionality).

andimik, 2023-10-20:
Thanks for the screenshot regarding the nice application 0x0093 the BBC is using :-)

andimik, 2023-10-20:
I confirm this bug. You will see `0.0` also when Method B is used. But in my case the layout (last line) is not broken.

And the `0.0` is visible also in the Advanced RDS view, not only in the AF/EON view.

mrwish7, 2023-10-20:
Thanks @andimik , yes the display of current frequency was something I added in the pull request linked above. I thought it was logical if AF sorting was disabled, so we can see the 'raw' list - inclusive of all frequencies.

I have made a ~30 minute MPX recording of BBC Radio 2, in case it's useful for anyone for this - https://mega.nz/file/AEJTXKQS#wBT_aPdLYav0eaXaB9F2yPSIfLhl4F_Wmm7HYHOhJ6Q

PE5PVB, 2023-10-23:
Fixed

PE5PVB, 2023-10-23:
BTW. If someone knows what data is transmitted with AID 0x0093, I like to hear this.

andimik, 2023-10-23:
https://www.etsi.org/deliver/etsi_en/301700_301799/301700/01.01.01_60/en_301700v010101p.pdf

You can find thd information there.

PE5PVB, 2023-10-23:
I can't find anything in redsea about calculating the DAB frequency. I checked 0x0093, it's linked to Group 12A. In group 12A I get A: C202 B: C144 C: 3717 D: CE15. How to convert this to the actual DAB frequency? @mrwish7 Can you tell me on which DAB frequency or channel this BBC radio is broadcasting in your area?

PE5PVB, 2023-10-23:
@mrwish7 I wonder if you can get more recording of British stations (not BBC national) who broadcast on DAB and have AID 0x0093.

PE5PVB, 2023-10-23:
I also love to have a recording of BBC Radio 4.

andimik, 2023-10-24:
https://github.com/windytan/redsea/blob/c6e6b47ac2c7a9aac9409483b00ca61cd6eb47bd/src/groups.cc#L1048

Here you find the table.

mrwish7, 2023-10-24:
Just catching up - thanks! Yes I can confirm that now there's no 0.0 value in the AF list in unsorted mode, however the list still starts from part-way through rather than the first value in the list being shown first (as shown in the RDS Spy screenshot).

I can confirm BBC National DAB (which has the ensemble ID CE15) is on 225.648 MHz - block 12B - 

[image]

The BBC also use different service IDs for their DAB versions compared with the respective RDS PI codes (PI code + 0x20), so that may also be mapped somewhere in the 0x0093 data.

I've just checked other stations I can receive here that are also on DAB (BBC local stations, Heart, Greatest Hits and Classic FM) but unfortunately none of these seem to be carrying group 0x0093 according to the TEF. Only Classic FM has an application flagged - RDS-TMC traffic info.

I will make an MPX recording of BBC Radio 4 for you.

PE5PVB, 2023-10-24:
In unsorted mode AF frequencies will be added as they are received. I can change this by waiting for the number of AF's first, but this won't be waterproof because the AF's are not numbered. So if one RDS packet was lost due what ever reason, the AF list is still not in the right order.

andimik, 2023-10-24:
> The BBC also use different service IDs for their DAB versions compared with the respective RDS PI codes (PI code + 0x20), so that may also be mapped somewhere in the 0x0093 data.

This is done in the DAB stream, you find it in FIG 0/6, but only repeated every 50 frames (as far as I found out)

```
$ etisnoop -i /tmp/bbc_04012019.eti -F 0/6 -n 50 | grep 0xC2
Adding FIG 0/6 to filter
Identified ETI type RAW
         DAB SId: 0xC225
         (AM-FM legacy): 0xC205
         DAB SId: 0xC221
         RDS PI: 0xC201
         DAB SId: 0xC222
         RDS PI: 0xC202
         DAB SId: 0xC223
         RDS PI: 0xC203
         DAB SId: 0xC224
         RDS PI: 0xC204
Decoded 50 ETI frames

```

I cannot check it actually (therefore I have opened a recording from 2019), as the BBC left 9 East some time ago and the UK spot beam is too far away from here.

But also in the http://tipok.org.ua/downloads/hardware/DAB-TX/Spartan6-FPGA/EasyDABv2/0,0x0425.bin.eti (from 2014) you find the same linking.

```
$ etisnoop -i /media/andreas/Elements/dab/0,0x0425.bin.eti -F 0/6 -n 50 | grep 0xC2
Adding FIG 0/6 to filter
Identified ETI type RAW
         DAB SId: 0xC224
         RDS PI: 0xC204
         DAB SId: 0xC225
         (AM-FM legacy): 0xC205
         DAB SId: 0xC221
         RDS PI: 0xC201
         DAB SId: 0xC222
         RDS PI: 0xC202
         DAB SId: 0xC223
         RDS PI: 0xC203
Decoded 50 ETI frames
```

mrwish7, 2023-10-24:
> In unsorted mode AF frequencies will be added as they are received. I can change this by waiting for the number of AF's first, but this won't be waterproof because the AF's are not numbered. So if one RDS packet was lost due what ever reason, the AF list is still not in the right order.

I think this would be OK, it generally worked quite well at this point (also including current frequency in unsorted AF) - https://github.com/PE5PVB/TEF6686_ESP32/pull/267

It can be useful to see the first frequency in the list for IDing purposes, personally I totally accept that if the signal isn't strong enough the list may be out of order/incorrect. I tried it out over the summer after it was first added and it actually worked a little better than I expected on some weaker signals, managing to load from the first value in the list at least.

I've made some MPX recordings for you -

An hour of BBC Radio 4 from a little earlier - https://mega.nz/file/tMhXSBpK#VLCwyZEYfmnBkjJvSXFOdU6KI-yYSoOscOpjIRsgHe8

And some other shorter clips of UK stuff, just in case it's useful. Classic FM includes RDS-TMC -

Classic - https://mega.nz/file/odIUDYJa#YA8fuZT-cnVdqhkfsx_7F-MVZHaVWFXWiFG17qtjxUg
Heart - https://mega.nz/file/EdpnyKhB#T6nGbT7V-RbQHFvEdW9KOKTg_1LqWMNRYVfmc7Yn8pE
Greatest Hits Radio - https://mega.nz/file/dZ4niKoA#XRHXoe_L_B0PWi-3z87v9jEhW395ZKdmeFhMSn2g9U4

PE5PVB, 2023-10-24:
`It's 0x3717 and you have to multiply by 0x10 (= add a 0) in order to get 0x37170, which is 225,648 MHz and block 12B, see https://www.fmlist.org/sendertabelle/dab-st_more.php?inputLand=G&inputEn=CE15 for instance.`

Yes, I also saw the trick with *10. But I was not sure if it was that simple. Reason why I ask, is it is an idea to add the DAB channel to the AF list ;-)

PE5PVB, 2023-10-26:
@mrwish7 Please confirm if the reported problem is fixed.

mrwish7, 2023-10-27:
Just checked now @PE5PVB - I can confirm that it's fixed. Thanks!

PE5PVB, 2023-10-29:
> `It's 0x3717 and you have to multiply by 0x10 (= add a 0) in order to get 0x37170, which is 225,648 MHz and block 12B, see [...]

Well, you have to multiply by 16 ;-)

PE5PVB, 2023-10-29:
I also added DAB AF in the AF/EON screen. Since the original problem is fixed, let's close this one.

closed 2023-10-29 by PE5PVB

andimik, 2023-10-30:
> Well, you have to multiply by 16 ;-)

Yes, I already told you, 0x10 is 16 (dec).

PE5PVB, 2023-10-30:
My bad ;-)