PE5PVB/TEF6686_ESP32 #346: Clear RDS data on changed PI-code

lxsxl, 2023-10-22:
I observed that in case of two stations on one frequency (both receivable with RDS by rotating the telescope) the RDS data won't be cleared if the PI-code changes. This can cause false RDS data; some examples:
1. Station 1 has RT+, station 2 doesn't: RT+ will permanently remain on the last received data from station 1, RT will sitch to the data of station 2.
2. Station 1 has EON, station 2 doesn't: EON from station 1 will permanently remain.
3. same for AF

I would suggest to adjust it like one of the following ideas:
1. Clear all RDS-data on PI-code change (probably except CT, see thread in discussions).
2. Exceptions may be useful in case of regional variants (switch only on second PI-nibble; to keep the received data alive when the network is temporarily separated into RV) AND/OR if the setting "Show RDS errors" is set to a higher level (let's say "Large" or "All") as loss of data might be unwanted in case of very poor reception.
3. If exceptions are too much effort there should be at least a setting for "Clear data on PI-change".

PE5PVB, 2023-10-23:
Problem is, how to correctly detect this. What happens when in one cycle the PI is received wrong. The error correction of RDS is not waterproof and can not be trusted.

lxsxl, 2023-10-23:
That's right. What about a error-acceptance percentage? So that at least those rarely errors won't cause clear of data? Or clear only if a new PI was detected let's say at least three times subsequently.

PE5PVB, 2023-10-23:
Only things that needs a reset are AF, RT+, EON, PTYN, AID and CT as far as I can see now.

lxsxl, 2023-10-23:
If DI, TP, PTY, ECC, LIC and PIN are updated continously I would agree.

PE5PVB, 2023-10-23:
I have to make a list what is updated and what not.

PE5PVB, 2023-10-24:
I made a script that checks several things and should do a RDS reset when PI is changed. Please test it.

lxsxl, 2023-10-24:
As I need a special place outside I have to test this tomorrow.

lxsxl, 2023-10-25:
After my test it works fine. IMHO you can close this issue.

closed 2023-10-26 by PE5PVB