PE5PVB/TEF6686_ESP32 #348: Problems with the new SPI speed function.

Overland-DX, 2023-10-23:
It looks like the new feature isn't working as it should. It detects the value set in User_Setup.h, but does not register desired values set in the menu.
Like for example when I set the value to 4MHz in user_setup.h, I get a noticeably slow radio. If I then set it to 60MHz in the menu, there is no difference. If I turn the radio off and on, it always goes back to the value set in user_setup.h.

ohmytime, 2023-10-24:
Restore defaults after shutting down, that's what I originally planned(Memory can be added if needed)
As for the SPI change not taking effect, I will investigate again.

PE5PVB, 2023-11-01:
Should be fixed. Please use the tft_espi library from ohmytime. https://github.com/ohmytime/TFT_eSPI_DynamicSpeed

andimik, 2023-11-01:
and don't forget to modify the TFT_eSPI configuration **again** like described in https://github.com/PE5PVB/TEF6686_ESP32/wiki/Installing-the-software

PE5PVB, 2023-11-02:
Is it all ok now? Then please close.

closed 2023-11-02 by Overland-DX