PE5PVB/TEF6686_ESP32 #350: AF method B detection

lxsxl, 2023-10-23:
I think that method B detection needs a bit adjustment. I received the following data:
[image]

Translated to AFs it is as follows:
#14 / 102,7
89,1 / 91,0
102,7 / 91,5
92,8 / 94,2
94,5 / 95,5
98,7 / 102,8
104,4 / 104,5
107,2 / filler

102,7 is < 50 % within this list and there is only one single list, so it should be detected as method A.

lxsxl, 2023-10-23:
And same for this:
[image]

This is:
#3 / 106,1
98,9 / 105,3

is also be method A (but detected as method B)

PE5PVB, 2023-10-23:
Is it correct that your radio is tuned to 102.7 in example 1 and to 106.1 in example 2?

lxsxl, 2023-10-24:
Yes, that's correct, sorry that I didn't mention it.

PE5PVB, 2023-10-24:
I've added an extra Method B probe. Please test if this is successfull.

lxsxl, 2023-10-24:
Everything I tested (including the examples above) was detected in the right way. Even method B with this crazy configuration (only number 1 and one frequency) is fine.

PE5PVB, 2023-10-24:
So this one can be closed?

lxsxl, 2023-10-24:
I would say YES

closed 2023-10-24 by PE5PVB