PE5PVB/TEF6686_ESP32 #354: RDS Block A, B and C wrong error calculation

andimik, 2023-10-24:
The error calculation for RDS blocks A, B and C is broken, it's similar to https://github.com/PE5PVB/TEF6686_ESP32/issues/189

I could repeat that several times.

[image]
[image]

For the hidden pixels on the left bottom corner I'll open a separate issue.

PE5PVB, 2023-10-24:
This this only on block A?

PE5PVB, 2023-10-24:
I modified the software so the "leds" are always written. For some reason they are sometimes not updated correct, while software says so. I guess a glitch in the compiler.

PE5PVB, 2023-10-26:
@andimik Please confirm if the problem is fixed.

andimik, 2023-10-29:
will test it in the next days.

lxsxl, 2023-11-01:
For me it is a bit strange that block A has almost no errors when the other blocks are shown with multiple errors. Or is there a technical reason that block A is less affected by errors?

PE5PVB, 2023-11-01:
Possibly because Block A is always the same.

lxsxl, 2023-11-01:
Of course, block A should remain the same. I thought that the LEDs display the error in that single block? So even if PI-code remains the same but a single block A was not completely received it should be a red LED, or am I wrong?

PE5PVB, 2023-11-01:
Don't know. Error data comes directly from the TEF chip. Possibly there is a kind of algorithm inside.

andimik, 2023-11-02:
For me it is solved, although sometimes group "LED" B, C, D are blinking very rapidly. But Rdsspy says BER = 0.

closed 2023-11-02 by andimik