PE5PVB/TEF6686_ESP32 #356: Abnormal word wrapping in menu selection screens.

MCelliotG, 2023-10-24:
The issue was brought up in the past and it's purely cosmetic. In the screens that appear after selecting an entry from the menus, the large font that is used gets wrapped abnormally, take a look at the following examples.

[image]
[image]
[image]
[image]

As you can see the last letter in the right edge is wrapping and the following one goes under it on the left, however the first letter of the entry gets completely hidden.
Now a workaround like cutting the words in translation _is not useful_ in this case, as in many languages the meaning of the entries can change dramatically. 
Does the Arduino code support custom wrapping after a certain and defined text length? I was searching myself for a solution but my code knowledge in Arduino is limited (unlike in Enigma2 where you can easily set custom wrapping). 
I only found this code, I don't know if it helps.

`String wrap(String s, int limit){

int space = 0;
 int i = 0;
 int line = 0;
 while(i<s.length()){
  
  if(s.substring(i,i+1)==" "){space=i; }
  if(line>limit-1){s=s.substring(0,space)+"~"+s.substring(space+1);line = 0;}
  i++;line++;}
 s.replace("~","\n");
  return s;
  }`

So, ideally the text should be set on where to start and where to end and after that it would wrap automatically on the next or previous available space, so it won't split the words randomly.

PE5PVB, 2023-10-24:
You can go to the second line using \n inside the text in translations. Like "Text1\nText2". Centre alligning will not work correct, so translaters should use spaces to fill the gaps. Better option is to keep the texts as short as possible.

Responsibility of text lengths is at the translators.

MCelliotG, 2023-10-24:
Even so, and shortening as I already said is not always an option, since not all languages are the same, the second entry should go lower, as it will overlap.

PE5PVB, 2023-10-24:
What will overlap?

MCelliotG, 2023-10-24:
When an entry is the same on both the small menu and the selection menu such as Show underscore in PS, if I put the/n in the translation it will wrap within the small menu and will cover up the value below it in the selection screen where the font is big.

MCelliotG, 2023-10-24:
I'll try to compile with /n in the translation and show what issues it causes.

PE5PVB, 2023-10-24:
\n not /n

MCelliotG, 2023-10-24:
Yes, you get what I mean. What I ask is if Arduino provides the option to set where a text begins and where it should wrap. I believe that should be the best way since in the menus with the large font there is no way to avoid wrapping no matter how short the entries will get.

PE5PVB, 2023-10-24:
I can make a script for that, other problem is that the same text is used in the menu overview, so what about that?

MCelliotG, 2023-10-24:
Maybe we can limit on what menus entries should wrap with a condition. I find it strange though that text manipulation is so limited in Arduino, at least in python there are many options with text size and wrapping.

PE5PVB, 2023-10-24:
It's not a limit in Arduino, but in the Library which converts things to pixels.

PE5PVB, 2023-10-26:
Question. Do the lines fit in the menu overview but not in the menu-open (bigger font) popup? If so, I can try to create a smaller font for the popups.

MCelliotG, 2023-10-26:
Yes, this is what is happening. Translators can manipulate the text in the menu entries to a certain point by shortening it, but in many languages there is a limit to how short entries can become without losing the meaning. In the popup the same entry will wrap with the big font. It wouldn't be such an issue if the text wrapped while the first word began at the left end of the screen, the problem is that the first letters go way too left and hide. If there was a way to make the text not hide on the left and wrap in the first available space instead of chopping the word, the problem would be solved completely.

MCelliotG, 2023-10-26:
[image]
This is an example of what is happening. If the text is set to begin at a certain point of a screen and ideally wrap in the space before the last word that reaches the right end of the screen it would be ideal.

PE5PVB, 2023-10-26:
I added textwrapping in the menubox.

When using \n in text, a second line is used in menubox. In normal menu overview the \n will be replaced by a space.

MCelliotG, 2023-10-26:
Let's try it!!!

closed 2023-10-27 by PE5PVB

MCelliotG, 2023-10-27:
I will do a bit of tiny adjustments on a couple of strings (Release button when ready etc) which should no go a bit lower and add the n to the languages I manage and we're good to go!!!