PE5PVB/TEF6686_ESP32 #357: FM double de-emphasis?

AaronJowitt, 2023-10-24:
build 181023. 

FM settings  > FM deemphasis: 50us
RDS settings > Region: Europe

Is FM audio de-emphasis being applied by both settings, resulting in unintended double de-emphasis?   On a quick listen, sounds better/normal with FM de-emphasis set to None.

PE5PVB, 2023-10-24:
Double de-emphasis is not possible.
I already removed the de-emphasis setting on RDS region last night.

AaronJowitt, 2023-10-24:
Thanks - makes sense the TEF wouldn't support that. I probably haven't listened long enough and suspected the wrong thing. So is;

FM settings > FM deemphasis: none

wrong?  When set to 'none' it sounds like there is some de-emphasis (maybe 50us?) still being applied.

AaronJowitt, 2023-10-24:
> Thanks - makes sense the TEF wouldn't support that. I probably haven't listened long enough and suspected the wrong thing. So is;
> 
> FM settings > FM deemphasis: none
> 
> wrong? When changed to 'none' it sounds like there is some de-emphasis (maybe 50us?) still being applied.

PE5PVB, 2023-10-24:
You are right, there was a bug in the de-emphasis. I've fixed that. Please check and report.

AaronJowitt, 2023-10-25:
Confirmed fixed in today's beta.  Thank you!

closed 2023-10-25 by PE5PVB