PE5PVB/TEF6686_ESP32 #365: Blinking PTYN

andimik, 2023-10-29:
This station from Croatia caused the PTYN field blinking, I have never seen that before.

[brac1.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/13199314/brac1.zip)

[VID_20231028_175029.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/fd19bd1c-8ffb-48d7-a0ae-5e779d99d2a7)

```
$ redsea -f brac1.wav -E | grep pty
{"pi":"0xC479","group":"10A","bler":4,"prog_type":"No PTY","pty_name":"        ","tp":false}
{"pi":"0xC479","group":"0A","bler":0,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"No PTY","ta":false,"tp":false}
{"pi":"0xC479","group":"10A","bler":6,"prog_type":"No PTY","pty_name":"        ","tp":false}
{"pi":"0xC479","group":"10A","bler":2,"prog_type":"No PTY","pty_name":"        ","tp":false}
```

Seems PTYN is just containing spaces.

Please note that I cannot test it again as this was during my vacation, therefore, please see the wav-file.

PE5PVB, 2023-10-29:
Yes. This station transmits PTY identification, but without any content, strange....:

Block 10A:
C479 A001 0000 0000
C479 A002 0000 0000

I changed the API so, when Block C and block D are empty, don't process PTYN.

andimik, 2023-10-29:
Well, I can only trust you have fixed it.

closed 2023-10-29 by andimik