PE5PVB/TEF6686_ESP32 #366: Missing PIN

andimik, 2023-10-29:
The following Croatian radio station has PIN (Program Item Number), but this is not shown on the TEF. It might even be the case that RT is not shown (but I cannot remember as I just took a photo).

Remark: for other channels it's ok.

[makarska.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/13199333/makarska.zip)

[image]

```
$ redsea -f makarska.wav -E | grep prog_item
{"pi":"0xC44D","group":"1B","bler":6,"prog_item_number":58163,"prog_item_started":{"day":28,"time":"12:51"},"prog_type":"Varied","tp":false}
{"pi":"0xC44D","group":"1B","bler":6,"prog_item_number":58163,"prog_item_started":{"day":28,"time":"12:51"},"prog_type":"Varied","tp":false}
{"pi":"0xC44D","group":"1B","bler":2,"prog_item_number":58163,"prog_item_started":{"day":28,"time":"12:51"},"prog_type":"Varied","tp":false}
{"pi":"0xC44D","group":"1B","bler":2,"prog_item_number":58163,"prog_item_started":{"day":28,"time":"12:51"},"prog_type":"Varied","tp":false}
```

[image]

Please note that I cannot test it as this was during my vacation.

It's https://github.com/PE5PVB/TEF6686_ESP32/commit/62ec6e06410ad411e4de0178fb442187a64e2401 (last version when I was at home).

PE5PVB, 2023-10-29:
Found it. This station was using group 1B for PIN. I changed it in the software.

PE5PVB, 2023-10-29:
Happy holidays.

andimik, 2023-10-29:
I'll trust you, so I close it.

closed 2023-10-29 by andimik