PE5PVB/TEF6686_ESP32 #368: CT has glitches

andimik, 2023-10-29:
Described initially by radiofan98 at https://discord.com/channels/1053804249651359765/1080501413651693578 , I also had this problem with beta from 28th Oct.

[image]

[image]

PE5PVB, 2023-10-30:
Do you have a step-by-step reproduction method? I can't figure it out.

lxsxl, 2023-11-01:
If a frequency gets weaker and RDS is lost, the time will switch to secondary colour but the font "shivers". And as long as RDS-reception won't return, CT stops running (shows permanently the last received CT-signal). If you switch to a different frequency and duration of RDS-loss is longer than one minute you will observe an overlyay.

PE5PVB, 2023-11-01:
OK. I'll check again. And yes, it's correct that the clock is frozen on RDS loss. It's ment to be.

PE5PVB, 2023-11-01:
Fixed. Please check.

andimik, 2023-11-01:
I just could read `1:61` due to reception error. I guess you should check if the time is plausible (same for PIN, where I also have seen >24:00)

PE5PVB, 2023-11-02:
I added extra checks

lxsxl, 2023-11-02:
Glitches seem to be gone. But what I don't understand: Why is the time stopped in case of poor reception but in case of tuning to a different station it will step up to the right time (even if there was no signal on the newly tuned frequency)? Imho it would be good to always fall back on the RTC. Thoughts of a layman, if this is too difficult to implement it can remain like it is.

PE5PVB, 2023-11-02:
Well I choose for this, because in my opinion when CT is detected on a service, it should show the values. When signal is lost, clock is not updated, so it stays that way. Maybe I'm a nerd on this ;-)

lxsxl, 2023-11-02:
Okay, I agree. What about this idea: If RDS is lost, keep the time running in secondary colour. If RDS returns (without tuning) also keep it running in secondary colour until the next CT-code is received. So it would be identical to the situation when you tune to a different frequency.

PE5PVB, 2023-11-02:
Yes, can do. Sorry for being a nerd ;-)

lxsxl, 2023-11-02:
You‘re perfect as you are 👍 on need to apologize

PE5PVB, 2023-11-07:
Done

lxsxl, 2023-11-07:
CT was received at 20:18, then I disconnected the antenna (remained on that frequency). Time switched as follows (of course in secondary colour): 20:19 - 20:20 - 20:19 - 20:20 - 20:21. Then I reconnected the antenna and on receiving RDS again the time went back to 20:18 (in primary colour).

PE5PVB, 2023-11-07:
fIXED.

lxsxl, 2023-11-08:
As far as I tested the time works well in all variants. For me this issue may be closed.

closed 2023-11-09 by PE5PVB