PE5PVB/TEF6686_ESP32 #372: Small bug - FrequencyColor parameter do nothing

zygmund2000, 2023-10-30:
Hi,

I'm trying to create a new color theme and I came across a definition that doesn't work in gui.cpp:
eg. `      FrequencyColor = 0x03DA;` doesn't work in any theme, color is the same as primary color.

PE5PVB, 2023-10-30:
You are right. It's a resedue from the past. I removed it. If you find any more unused color references, please let me know.

closed 2023-10-30 by PE5PVB

zygmund2000, 2023-10-30:
It was nice to change this color, why can't it be like before?

PE5PVB, 2023-10-30:
Why should you? Too much different colors can make you crazy like a clown ;-)

PE5PVB, 2023-10-30:
bde1331#r131296671

zygmund2000, 2023-10-30:
So why not? I don't need colors like clown or LGBT, but colors always are nice

zygmund2000, 2023-10-30:
I've changed 1 line:
`          FrequencySprite.setTextColor(FrequencyColor, PrimaryColorSmooth, false);`
And now it seems to look as before.