PE5PVB/TEF6686_ESP32 #376: PI-Code with "show PI-errors" = on

lxsxl, 2023-11-01:
If the setting "show PI errors" is switched on, it often happens that the PI-code "shivers". It looks like primary and either secondary or black colour toggle very fast. It remains even if RDS-reception gets lost. This happens typically in case of poor reception.

Also it seems that the questionmarks are not cleared when an uncertain reception lasted a few seconds (and even if RDS-reception gets better later).

PE5PVB, 2023-11-01:
Fixed.

PE5PVB, 2023-11-02:
@lxsxl Please confirm and close.

lxsxl, 2023-11-02:
At least the PI-code won't get stuck anymore. But the "shivering" is still occuring, typically near the threshold between "no RDS" and "RDS available". Maybe this is intended? As a further step to one or two questionmarks?

PE5PVB, 2023-11-02:
I guess the radio is switching so fast between ? and ?? that it looks like shivering. It rewrites the whole PI with questionmarks, because it can't know the exact x-offset because the fontwidth is different per character.

lxsxl, 2023-11-02:
https://github.com/PE5PVB/TEF6686_ESP32/assets/140750201/af2e137d-04ba-4a0a-90c8-89eee7e1cee6

Maybe this makes it a bit clearer

PE5PVB, 2023-11-02:
Can you try this again with RDS buffer on?

lxsxl, 2023-11-02:
What is „RDS buffer“? I tried with „RT buffer“ on but it‘s the same behaviour.

PE5PVB, 2023-11-02:
When switched off the rds blocks are send from TEF chip without any kind of sync. You can see the effect on the error leds, they will flash very quick after a tune. 

Pro: very vast RDS decoding after tuning
Con: Sync issue, see your video ;-)

lxsxl, 2023-11-02:
Okay, but which setting do I have to adjust? Or is it within „RT buffer“?

PE5PVB, 2023-11-02:
RT Buffer is something different. This buffers for example RT A and pushes it when RT B becomes active, and so on. This will prvent showing you a building RT

lxsxl, 2023-11-02:
Okay. But what is „RDS“? I don‘t find it in the menu. Or is it „RDS filter“?

PE5PVB, 2023-11-02:
Yes. Sorry

lxsxl, 2023-11-02:
It is still the same.

PE5PVB, 2023-11-07:
I added an extra check, hopefully it is better now. Still, during PI scan in error mode it can be set to different values. This goes very quick and for you it looks like it's flashing.

lxsxl, 2023-11-07:
Actually it seems to be fine, I couldn't see any flashing. So I close this issue.

closed 2023-11-07 by lxsxl