PE5PVB/TEF6686_ESP32 #381: AF/EON show red color instead of green at 2nd or 3rd scan

andimik, 2023-11-02:
When you are in advanced RDS view and go the first time to AF/EON view the colors are correct.

But if you repeat that **without** changing the frequency green frequencies might become red.

PE5PVB, 2023-11-02:
When you repeat this over and over again will they sometimes end up green?
Is this station possibly somewhat weak?

andimik, 2023-11-02:
No, I expect to have it written in green on all my attempts, as this is my local transmitter (>60 dBuV).

Have repeated it with other stations as well, same result.

If it would be too week, the color will be yellow, but not red.

PE5PVB, 2023-11-02:
OK, does it always stay red or sometimes green after several attempts?

andimik, 2023-11-02:
Good question, I just repeated it around three times for each frequency. I will try more attempts.

PE5PVB, 2023-11-02:
Should be fixed.

lxsxl, 2023-11-02:
In my last tests it seemed to work fine.

andimik, 2023-11-03:
Yes, looks good, have not seen that again. Therefore closing.

closed 2023-11-03 by andimik