PE5PVB/TEF6686_ESP32 #386: RDS logo reacts very slow

lxsxl, 2023-11-03:
Idk if you are still working on this but actually the RDS logo reacts very slow: RDS data are already received when RDS logo is still inactive and also vice versa, when RDS get's lost the logo will remain until there is almost nothing to be heard anymore.

PE5PVB, 2023-11-03:
I can't reproduce that

lxsxl, 2023-11-04:
Uploading IMG_8503.mov…

second video will follow…

lxsxl, 2023-11-04:
https://github.com/PE5PVB/TEF6686_ESP32/assets/140750201/c8ddf30e-d493-450f-9a7b-70890e508dc4

lxsxl, 2023-11-04:
[image]

those are the RDS settings. In the first video you can see disconnecting the antenna what causes loss of reception but RDS remains active and everything is in primary colour. The second video is vice versa (reconnecting antenna and you can see that RT is updated though RDS indicator is off. (Not recorded in the videos: If I tune up and down to the same frequency RDS will be immediately active).

Hope I found the reason: If I turn off „RDS-Filter“ it works fine, so likely there will be a bug.

PE5PVB, 2023-11-07:
Fixed.

lxsxl, 2023-11-07:
I confirm that it works fine. Close this issue as completed.

closed 2023-11-07 by lxsxl