PE5PVB/TEF6686_ESP32 #387: Support for 2nd EON frequency in case of multiple tuning frequency

andimik, 2023-11-03:
I've realized that the EON Frequency might not be correct as the tuned frequency is used more than once.

[image]

In my example I have tuned to 94.7 (RADIO-K is described as OE2 K in the below screenshots) which is used twice here. So the EON frequency for OE 1 can be either 87.7 or 89.7. 

I've checked the official frequency table

[image]

[image]

In RDS, both are mentioned:

```
{"pi":"0xA502","group":"14A","other_network":{"pi":"0xA201","kilohertz":89700,"tp":false},"prog_type":"No PTY","tp":true}
{"pi":"0xA502","group":"14A","other_network":{"pi":"0xA201","kilohertz":87700,"tp":false},"prog_type":"No PTY","tp":true}
```

So I propose that both possible frequencies are shown alternating as this is not unique.

PE5PVB, 2023-11-03:
So what is your suggestion?

andimik, 2023-11-04:
Write a second line with the frequency and in front the word `or`

```
A201   OE 1   87.7
             or     89.7
```

lxsxl, 2023-11-04:
Did you consider that we already added (at total) three EON-frequencies on advanced RDS-screen to resolve this problem?
Anyway if more than one EON-frequency shall be shown also on EON/AF-screen it would be a further reason to separate AF and EON screens.

andimik, 2023-11-05:
Oh, I didn't know that. I was never using the Advanced RDS view for AF and EON.

PE5PVB, 2023-11-07:
Can you open a topic in Discussions about this? This is a feature request.

closed 2023-11-07 by PE5PVB