PE5PVB/TEF6686_ESP32 #394: Tickers slow down on RDS loss

lxsxl, 2023-11-07:
Tickers on both screens slow down if RDS reception is lost. They speed up if RDS is received again.

lxsxl, 2023-11-07:
This is not new but until now I forgot to post it.

PE5PVB, 2023-11-07:
Fixed.

lxsxl, 2023-11-08:
Actually the behaviour is still the same (code from this night).

andimik, 2023-11-13:
> Fixed.

No, not fixed, it's still the case in 95cbfe98d51c8dfc

More, the text `No RT+ available` suddenly starts scrolling after signal loss, which will **not** scroll during reception.

[VID_20231113_210755.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/f206e639-0786-4f9f-b083-f4461c4391e2)

BTW: What about `ECC`, `LIC`, `PTYN`, `PIN` after signal loss? I expect that their color change to blue as well.

PE5PVB, 2023-11-14:
Hi,

I did some investigation on this. It seems that your problem is triggered by the low level threshold. If signal drops below the set point, screenupdates are slowed down to avoid extreme noise. Try it by setting the value to -15dB.

lxsxl, 2023-11-14:
With this setting the scrolling remains in the usual speed. But if I activate AF-tuning, it will stop each (approx.) second for a few milliseconds. It seems that this short stop is the moment where the TEF tries to tune to an AF. But I would say it is acceptable. If I keep the low level threshold on -15dB does it affect any negative behavior?

andimik, 2023-11-14:
Why not set it to -15dB by default?

PE5PVB, 2023-11-14:
Yes, I've changed the default setting this afternoon.

lxsxl, 2023-11-14:
Okay, then I close this issue.

closed 2023-11-14 by lxsxl