PE5PVB/TEF6686_ESP32 #395: AF-B mode is not displayed

Yurij20, 2023-11-08:
In the latest update, the AF-B mode is no longer displayed.  Also, there are no S and R marks in the AF frequency list

PE5PVB, 2023-11-08:
I can't reproduce it. Are you sure the radiostation uses AF Method B? Did you wait long enough, it can take some time to detect AF-B.

Yurij20, 2023-11-08:
In the release for example from 10/25/2023 on 104.2 and 90.3, for example, AF-B is immediately displayed, in the latter the AF-B method is not visible at all.

PE5PVB, 2023-11-08:
In that version was a bug which false identified some channels as AF-B. I think the service you are checking doesn't have AF-B.

Yurij20, 2023-11-08:
Now we tested it with friends at one of the radio stations, specifying the af-b method, and on the latest firmware it doesn’t show up at all, alas. Below is the photo
[image]
[image]
[image]

PE5PVB, 2023-11-08:
Can you check with RDS Spy?

Yurij20, 2023-11-08:
[image]

PE5PVB, 2023-11-08:
It seems that this transmitter is using the AF-B the wrong way. To work correctly it should also send the same RDS data on the three other frequencies, and you should have more than 1 AF-B list. To prevent a wrong AF-B detection I added a check that at least one AF-B list for the current frequency is detected and one for another frequency is detected.

If I drop this, it can end up detecting false AF-B on some stations. Only alternative is to count the AF's which are the same after each listing, but that is a lot of work and can give faulty information on weak signals.

Yurij20, 2023-11-08:
Here is the situation with three af-b lists.  still doesn't display
[image]
[image]
[image]

PE5PVB, 2023-11-08:
Can you get me a MPX recording?

Yurij20, 2023-11-08:
ok) I’ll do it when I’m in another apartment, there’s an opportunity to record it.

Yurij20, 2023-11-09:
By the way, I have an MPX recording of one of our radio stations using the AF-B method.  This method also stopped being displayed on this station, although it was displayed perfectly in earlier builds.  I sent it to you earlier on another issue.  Check it out, if it doesn’t work then I’ll write another one later)

https://disk.yandex.ru/d/iMA0cRtA9pLCpA

PE5PVB, 2023-11-09:
From which frequency was this recorded? 101.5?

Yurij20, 2023-11-09:
102.5

PE5PVB, 2023-11-09:
Should be fixed.

Yurij20, 2023-11-09:
ok) I'm waiting for the firmware release, I'll check it

Yurij20, 2023-11-10:
checked, everything works thanks

closed 2023-11-10 by Yurij20