PE5PVB/TEF6686_ESP32 #398: AF-method B is not detected if frequency is not tuned exactly

lxsxl, 2023-11-09:
If you tune the frequency a few 10 kHz lower or higher than full 100 kHz, AF-method B will not be detected and therefore also R/S are not indicated. This should likely be because the AF-head-frequencies will be compared with the tuned frequency (and in this setting it won't match). I have two different ideas how this might be fixed:

1. Round the tuned frequency (e. g. 89,95 to 90,04 will be interpretated as 90,0). This solution should be easy but will not fit in 100 %.
2. The second idea should be perfect but likely more complicated: Only to detect AF-method B it is NOT necessary to compare the head-frequencies with the tuned frequency (of course this is needed later to assign the connected EON-frequencies and/or the R/S-status). There are rather two facts that indicate AF-method B: a) there are two or more head-frequencies (each leaded by a number-code) within the AF-"stream" OR b) if there is only one head-frequency, this is transmitted in each AF-pair. (special case: if there is only one single AF transmitted, it should be handled as AF-method A though subpoint b) would be true).

A scond point in this topic: In case that the tuned frequency is a bit lower or higher than full 100 kHz the correct tuned frequency will also be shown as an AF. If you tune the correct frequency it will not be visible as an AF (in case of only one single AF the list remains empty). If the tuned frequency is transmitted as AF it should then also be visible in any case (hope this doesn't affect problems with AF-tuning).

PE5PVB, 2023-11-09:
Hi,

1. Good idea. I changed it in the API.
2. I want to count the times the currentfreq is broadcasted in every AF cycle. When more than 1, it should be AF-B. Will try to create this tonight.

PE5PVB, 2023-11-09:
Point 2. Done.

lxsxl, 2023-11-09:
Both seem to work quite fine :-)
What about displaying also the tuned frequency itself if it is transmitted in the AF-list?

PE5PVB, 2023-11-09:
I remove the tuned frequency to have one extra AF space in the memory.

lxsxl, 2023-11-09:
Okay, then I will close this issue as completed.

closed 2023-11-09 by lxsxl