PE5PVB/TEF6686_ESP32 #426: Improvement of AF-method B detection

lxsxl, 2023-11-18:
We should improve AF-method B detection a bit. Actually method B is indicated if the tuned frequency is transmitted more than once in a list. This may lead to a wrong indication. For example: "Dlf" on 102,7 transmits 14 AFs in one single list, where the tuned frequency is included twice. So actually the TEF indicates this as AF-method B. 

Though this of course a small error by the broadcaster (as each frequency should be only once in method A), I suggest to modify the detection a bit:

1. If the head-frequency (the one which is in pair with the [number of AFs]) is transmitted in **each** AF-pair of this list (>= 50 % of a single AF-list), method B has to be indicated. Even if further lists for different head-frequencies will follow, the indicator can be activated after the first list is received.
2. It may happen that only the first two bytes of an AF-list ([number of AFs] and [head frequency]) were received and the rest of this list could not be received completely. In that case when a second (different) pair of [number of AFs] and [head frequency] is received, AF-method B has also to be indicated (as two different list are not possible in method A).

PE5PVB, 2023-11-18:
What if I change it to activate it when the currentfreq is detected more than 2 times? It's difficult to find workarounds when transmitters are not correctly configured.

lxsxl, 2023-11-18:
This could be an easy solution. It would always work except that a method-B AF-list contains only one single AF. But I think this case will be extremely unlikely.

lxsxl, 2023-11-20:
Some further thoughts to this issue:
1. Am I right how things work: a) method B is indicated if in one AF-list (no matter which one) the head-frequency is used as AF at least once AND b) R/S-indicators in AF-screen will only be displayed if all (!) AF in the AF-list of the tuned frequency are pairwise with the tuned frequency?
2. If this is right, then you should not change anything. Both information together is a good combination: extremely fast method B-detection (though with a slightly potential error if the broadcaster doesn’t has a perfect list) and a correction of this error in the AF-list as this will remain a method A-list (as it doesn‘t have any R/S-indicators.
3. If this is wrong, please help me how the algorithms actually work.

PE5PVB, 2023-11-21:
Current flow:

1. When a 'number of following AF's' is detected, AF flag is enabled.
2. When a 'number of following AF's' and a frequency which is not the current frequency is detected, method B trigger is set.
3. When trigger is set and 'number of following AF's' and the current frequency is detected, Method B is enabled... OR when the current frequency is detected more than once in a AF cycle (from number of following AF's to number of following AF's), Method B is enabled.

So there are two trigger points to enable method B.

lxsxl, 2023-11-21:
Three questions:
1. "When a 'number of following AF's' and a frequency which is not the current frequency" means the frequency in pair withe the "number of..." or anywhere in the list?
2. "When trigger is set and 'number of following AF's' and the current frequency" - same question as above
3. In my special case in the initializing post we have the following AFs: #14 / 102,7 -- 89,1 / 91,0 -- 102,7 / 91,5 -- 92,8 / 94,2 -- 94,5 / 95,5 -- 98,7 / 102,8 -- 104,4 / 104,5 -- 107,2 / filler
In this case AF-B **is** activated but there is no R-indicator (what could have been expected for 91,5; **but please, keep like it is, because this is not an AF-B list but only a slightly faulty AF-A list**). Though the question is, when are R-/S-indicators activated? Only when there are at least two AF-lists? Or when the list (or all lists) have the head-frequency in each AF-pair?

lxsxl, 2023-11-21:
Seems that it depends on the order the AFs are received because in a different run, the 91,5 get's the "R"-indicator.

PE5PVB, 2023-11-21:
R/S indicators are marked, even when method B is not active (yet). But only when the number of AF's is paired with currentfreq.

lxsxl, 2023-11-21:
But in my example AF-B is detected and number of AFs is paired with current freq. But in some attempts there will be no "R", in other attempts it will be shown.

lxsxl, 2023-11-21:
[image]
[image]

lxsxl, 2023-11-21:
I waited really long each time

PE5PVB, 2023-11-21:
Can you send me a stream and the frequency?

lxsxl, 2023-11-21:
[D210 Dlf 102,7.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/13431241/D210.Dlf.102.7.zip)

PE5PVB, 2023-11-21:
I think method B is false detected by RDS Spy because of double 102.7 in the listing.

andimik, 2023-11-22:
Let me check in `redsea`

lxsxl, 2023-11-22:
The AF sequence is like I wrote above. And due to the double 102,7 it is quite logical that AF-B is detected (though it is AF-A).
But why is the 91,5 mostly without “R” and rarely with this indication?
Could the moment of hitting the RDS stream be the reason?

PE5PVB, 2023-11-22:
Is it false detected as method B by the TEF?

andimik, 2023-11-22:
I've downloaded the wave file and checked on my TEF, it's method A, which is what `redsea` also says.

```
$ redsea -r 192k -f ./D210\ Dlf\ 102\,7.wav | sort | uniq
{"pi":"0xD210"}
{"pi":"0xD210","group":"0A","di":{"artificial_head":false},"is_music":true,"prog_type":"Serious classical","ta":false,"tp":false}
{"pi":"0xD210","group":"0A","di":{"compressed":false},"is_music":true,"prog_type":"Serious classical","ta":false,"tp":false}
{"pi":"0xD210","group":"0A","di":{"dynamic_pty":true},"is_music":true,"prog_type":"Serious classical","ta":false,"tp":false}
{"pi":"0xD210","group":"0A","di":{"stereo":true},"is_music":true,"prog_type":"Serious classical","ta":false,"tp":false}
{"pi":"0xD210","group":"0A","ps":"Dlf     ","alt_frequencies_a":[89100,91000,91500,92800,94200,94500,95500,98700,102700,102700,102800,104400,104500,107200],"di":{"stereo":true},"is_music":true,"prog_type":"Serious classical","ta":false,"tp":false}
{"pi":"0xD210","group":"0A","ps":"Dlf     ","di":{"stereo":true},"is_music":true,"prog_type":"Serious classical","ta":false,"tp":false}
{"pi":"0xD210","group":"14A","other_network":{"pi":"0xD220","alt_frequencies":[98900,100200,106000,90500,89000,96500,98900],"tp":false},"prog_type":"Serious classical","tp":false}
{"pi":"0xD210","group":"14A","other_network":{"pi":"0xD220","has_linkage":false,"tp":false},"prog_type":"Serious classical","tp":false}
{"pi":"0xD210","group":"14A","other_network":{"pi":"0xD220","prog_type":"No PTY","ta":false,"tp":false},"prog_type":"Serious classical","tp":false}
{"pi":"0xD210","group":"14A","other_network":{"pi":"0xD220","ps":"DKULTUR ","tp":false},"prog_type":"Serious classical","tp":false}
{"pi":"0xD210","group":"14A","other_network":{"pi":"0xD220","tp":false},"prog_type":"Serious classical","tp":false}
{"pi":"0xD210","group":"2A","prog_type":"Serious classical","tp":false}
{"pi":"0xD210","group":"2A","radiotext":"Das macht die Welt so tiefschön.Ästhetik des Todes in Liedern d","prog_type":"Serious classical","tp":false}
{"pi":"0xD210","group":"2A","radiotext":"Deutschlandfunk - Alles von Relevanz","prog_type":"Serious classical","tp":false}
{"pi":"0xD210","group":"2A","radiotext":"Musikszene","prog_type":"Serious classical","tp":false}
{"pi":"0xD210","group":"4A","clock_time":"2023-11-21T22:44:00+01:00","prog_type":"Serious classical","tp":false}
```

PE5PVB, 2023-11-22:
So now ok?

PE5PVB, 2023-11-22:
So now ok?

andimik, 2023-11-23:
> Is it false detected as method B by the TEF?

No, I have tested the file in my lab, TEF says method A.

PE5PVB, 2023-11-23:
So this one can be closed?

lxsxl, 2023-11-23:
It is still detected as method B. But 91,5 is now always indicated as „R“. So far the algorithms are now working properly as you described them above.

Method is wrong anyway in this case (as andimik reports by redsea diagnosis).

The algorithm still should be adjusted, but as this is a really special issue it can be planned for a future patch if it is too much actually.

lxsxl, 2023-11-23:
The needed adjustment would be the following (regarding step 3):
Actually it is
When trigger is set and 'number of following AF's' and the current frequency is detected, Method B is enabled... OR when the current frequency is detected **more than once** in a AF cycle (from number of following AF's to number of following AF's), Method B is enabled.

and it should be:
When trigger is set and 'number of following AF's' and the current frequency is detected, Method B is enabled... OR when the current frequency is detected **>= 50 %** in a AF cycle (from number of following AF's to number of following AF's), Method B is enabled.

andimik, 2023-11-23:
> It is still detected as method B

On Dlf? 

See above, the same file you were sharing lets TEF display Method A, not B.

lxsxl, 2023-11-23:
That‘s really strange:
If I tune to Dlf/102,7 and switch to advanced RDS screen, it only shows „AF“. Then switch to AF screen, the list is without any „R“ indicator. Back to advanced RDS, it is still only „AF“. Again to AF screen, after another AF-cycle there appears the „R“ for 91,5, then back to advanced RDS mode and you have „AF-B“.
😮

lxsxl, 2023-11-23:
Another attempt: I restarted the TEF and now it shows always „AF-B“ without any „in and out“. I‘m totally confused…

lxsxl, 2023-11-23:
And again: There is no guarantee to reproduce it safely. Sometimes it detects AF-B directly, sometimes you need one, two or three tries of “in and out” until AF-B is detected.

PE5PVB, 2023-11-23:
```
When trigger is set and 'number of following AF's' and the current frequency is detected, Method B is enabled... OR when the current frequency is detected >= 50 % in a AF cycle (from number of following AF's to number of following AF's), Method B is enabled.
```

I've changed this. But I was not able to reproduce your problem with the DLF recording I received.

lxsxl, 2023-11-23:
I again compared the algorithm with the AF sequence: Step 2 of the algorithm is NOT true in this case (“When a 'number of following AF's' and a frequency which is not the current frequency is detected, method B trigger is set.”) The ‘number of following AFs’ is paired with the current frequency (and not a different AF). Maybe there is a problem? Or may the not current frequency be anywhere in the list?

lxsxl, 2023-11-23:
Oh sorry, this was overlapping. I will try with the new code. Will let you know

PE5PVB, 2023-11-23:
> Oh sorry, this was overlapping. I will try with the new code. Will let you know

It doesn't matter, because if it's not paired with currentfreq, the currentfreq will not be mentioned that much in that list.

lxsxl, 2023-11-23:
For Dlf 102,7 it works now. The detection remains on "AF".

But we still have a (hopely very last) small incorrectness with the 50 % (sorry, my fault): It is the frequency in pair with 'number of AFs' that must be >= 50 % in the same list (and not any frequency). For this I've got another example (D220 Dlf_Kult on 106,1): The sequence is as follows: '#3 / 106,1 -- 98,9 / 105,3. Actually this is detected as method B (but of course is method A). I will record an MPX stream and send it in a few minutes.

lxsxl, 2023-11-23:
[D220 Dlf Kult 106,1.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/13453291/D220.Dlf.Kult.106.1.zip)

PE5PVB, 2023-11-23:
OK. try again

lxsxl, 2023-11-23:
I'm sorry, just broke my device. 5 V-wire is disconnected from the board, so no connection to arduino IDE. I will ask on discord if someone can help ...

lxsxl, 2023-11-23:
Fixed it to do at least the flashing. AF method detection is perfect now. Thanks for your patience! Now I need to get a micro soldering iron to fix the wiring durable. There is not much capacity left ...

lxsxl, 2023-11-23:
I will close this issue as completed.

closed 2023-11-23 by lxsxl