PE5PVB/TEF6686_ESP32 #427: EON-PS improvement

lxsxl, 2023-11-18:
In case of poor reception, the EON-PS (or at least the previously received parts of it) is often cleared in the next EON-PS-cycle if the respective dataset is not (fully) received. It would be nice, if the (in previous cycles) safely received parts of the EON-PS will remain. Only if a different EON-PS was received safely of course it should be updated. I estimate that this behaviour is already typical for the other EON-parts.

PE5PVB, 2023-11-18:
This is not possible, because the length of the PS may vary.

lxsxl, 2023-11-18:
I think you misunderstood me. I will make an example:

If reception is good, WDR 2 (D392) transmits EON-data for 1LIVE (D391). Group 14A will transmit as follows:

variant-code 0001 -> block 3 PS char. 1 & 2 "1L" / block 4 PI D391
variant-code 0010 -> block 3 PS char. 3 & 4 "IV" / block 4 PI D391
variant-code 0011 -> block 3 PS char. 5 & 6 "E_" / block 4 PI D391
variant-code 0100 -> block 3 PS char. 7 & 8 "__" / block 4 PI D391
=> After this one cycle PS will be completely displayed (1LIVE___).

If reception is bad it actually works like this (a few cycles): 

variant-code 0001 -> block 3 PS char. 1 & 2 *not received* / block 4 PI D391
variant-code 0010 -> block 3 PS char. 3 & 4 "IV" / block 4 PI D391
variant-code 0011 -> block 3 PS char. 5 & 6 *not received* / block 4 PI D391
variant-code 0100 -> block 3 PS char. 7 & 8 "__" / block 4 PI D391
=> PS will be displayed as "__IV____

next cycle without any received PS-characters => PS will change to "________"

next cycle:
variant-code 0001 -> block 3 PS char. 1 & 2 "1L" / block 4 PI D391
variant-code 0010 -> block 3 PS char. 3 & 4 *not received* / block 4 PI D391
variant-code 0011 -> block 3 PS char. 5 & 6 "E_" / block 4 PI D391
variant-code 0100 -> block 3 PS char. 7 & 8 *not received* / block 4 PI D391
=> PS will change to ""1L__E___"

next cycle without any received PS-characters => PS will change to "________"

You could improve this as follows:

variant-code 0001 -> block 3 PS char. 1 & 2 *not received* / block 4 PI D391
variant-code 0010 -> block 3 PS char. 3 & 4 "IV" / block 4 PI D391
variant-code 0011 -> block 3 PS char. 5 & 6 *not received* / block 4 PI D391
variant-code 0100 -> block 3 PS char. 7 & 8 "__" / block 4 PI D391
=> PS will be displayed as "__IV____"

next cycle without any received PS-characters => PS remains "__IV____"

next cycle:
variant-code 0001 -> block 3 PS char. 1 & 2 "1L" / block 4 PI D391
variant-code 0010 -> block 3 PS char. 3 & 4 *not received* / block 4 PI D391
variant-code 0011 -> block 3 PS char. 5 & 6 "E_" / block 4 PI D391
variant-code 0100 -> block 3 PS char. 7 & 8 *not received* / block 4 PI D391
=> PS will change to "1LIVE___"

next cycle without any received PS-characters => PS remains "1LIVE___"

lxsxl, 2023-11-18:
sorry some underscores were deleted (formatting problem?) I'll try again...

lxsxl, 2023-11-18:
(I replaced the underscores by pipes)

If reception is good, WDR 2 (D392) transmits EON-data for 1LIVE (D391). Group 14A will transmit as follows:

variant-code 0001 -> block 3 PS char. 1 & 2 "1L" / block 4 PI D391
variant-code 0010 -> block 3 PS char. 3 & 4 "IV" / block 4 PI D391
variant-code 0011 -> block 3 PS char. 5 & 6 "E|" / block 4 PI D391
variant-code 0100 -> block 3 PS char. 7 & 8 "||" / block 4 PI D391
=> After this one cycle PS will be completely displayed (1LIVE|||).

If reception is bad it actually works like this (a few cycles):

variant-code 0001 -> block 3 PS char. 1 & 2 not received / block 4 PI D391
variant-code 0010 -> block 3 PS char. 3 & 4 "IV" / block 4 PI D391
variant-code 0011 -> block 3 PS char. 5 & 6 not received / block 4 PI D391
variant-code 0100 -> block 3 PS char. 7 & 8 "||" / block 4 PI D391
=> PS will be displayed as "||IV||||"

next cycle without any received PS-characters => PS will change to "||||||||"

next cycle:
variant-code 0001 -> block 3 PS char. 1 & 2 "1L" / block 4 PI D391
variant-code 0010 -> block 3 PS char. 3 & 4 not received / block 4 PI D391
variant-code 0011 -> block 3 PS char. 5 & 6 "E|" / block 4 PI D391
variant-code 0100 -> block 3 PS char. 7 & 8 not received / block 4 PI D391
=> PS will change to ""1L||E|||"

next cycle without any received PS-characters => PS will change to "||||||||"

You could improve this as follows:

variant-code 0001 -> block 3 PS char. 1 & 2 not received / block 4 PI D391
variant-code 0010 -> block 3 PS char. 3 & 4 "IV" / block 4 PI D391
variant-code 0011 -> block 3 PS char. 5 & 6 not received / block 4 PI D391
variant-code 0100 -> block 3 PS char. 7 & 8 "||" / block 4 PI D391
=> PS will be displayed as "||IV||||"

next cycle without any received PS-characters => PS remains "||IV|||"

next cycle:
variant-code 0001 -> block 3 PS char. 1 & 2 "1L" / block 4 PI D391
variant-code 0010 -> block 3 PS char. 3 & 4 not received / block 4 PI D391
variant-code 0011 -> block 3 PS char. 5 & 6 "E|" / block 4 PI D391
variant-code 0100 -> block 3 PS char. 7 & 8 not received / block 4 PI D391
=> PS will change to "1LIVE|||"

next cycle without any received PS-characters => PS remains "1LIVE|||"

PE5PVB, 2023-11-18:
Yes, but what about dynamic PS? That will be one mess.

lxsxl, 2023-11-18:
I have never seen dynamic EON-PS. As EON-PS is transmitted much less than real PS it wouldn't make much sense. But of course, if a broadcaster would use it dynamically it would be a mess. I tested with several stations from RTBF: All networks use dynamic PS but all EON-PS are static. I can't receive any german station with dynamic PS but I suppose that it is the same.

PE5PVB, 2023-11-18:
Well, here we have only dynamic PS on EON. I tried it your way and this ends up with a mess on PS.

lxsxl, 2023-11-18:
Okay, that's a pity. What about a setting (EON-PS static / EON-PS dynamic)?

PE5PVB, 2023-11-19:
Maybe an idea for the next update. I want to close in to a release.

lxsxl, 2023-11-19:
Okay, I will close this issue and move it to the discussion-board.

closed 2023-11-19 by lxsxl