PE5PVB/TEF6686_ESP32 #434: Primary colour in case of RDS loss

lxsxl, 2023-11-19:
If RDS get's lost and you return from AF/EON-screens to advanced RDS-screen PS, PI and PTY will be in primary colour for a few milliseconds and then switch to secondary colour.

andimik, 2023-11-19:
At what stage RDS needs to get lost? 

Because I tried but could not get what your are writing.

andimik, 2023-11-19:
Ah, switching 2x from normal to advanced and AF view and back will print PS and PI in yellow, but the rest in blue. You are right, the colors are wrong then.

PE5PVB, 2023-11-21:
Fixed.

lxsxl, 2023-11-21:
I close the issue as completed.

closed 2023-11-21 by lxsxl

andimik, 2023-11-22:
No, PTYN will stay in yellow. Please reopen it.

andimik, 2023-11-22:
No, PTYN is still in wrong color. Seems in https://github.com/PE5PVB/TEF6686_ESP32/commit/c2d89bab245419871dbb8b7d6bc2646f033876b7 there is no fix for ptynold

[image]

What about ECC/LIC/PIN color?