PE5PVB/TEF6686_ESP32 #447: Flickering in RDS-block error-dots

lxsxl, 2023-11-23:
I repeat this message from discord chat:

If you return from menu to main screen and then switch to advanced RDS mode the block error dots are flickering for a few seconds and after that they are blinking in the normal frequency. The longer you stayed in the menu the longer the flickering is visible.

https://github.com/PE5PVB/TEF6686_ESP32/assets/140750201/be5be00b-5b15-4d3d-9924-7d3faf5ec28c

PE5PVB, 2023-11-23:
Yes, this is correct. When you switch on RDS filter in the menu you don't have this problem. The effect you see is caused by empty RDS buffers.

Yurij20, 2023-11-23:
By the way, since the topic has started, I’ll say about the connection between the situation and the PS line :) Previously, I wrote everything about PS errors with a weak signal, but recently I noticed that just at the moment of rapid flickering of the PS indicators, the PS line works perfectly with weak reception, as in old firmware.  As soon as the flickering stops everything comes back with errors.  So, in order to now clearly read the PS line, I call up the menu for a certain time and exit it and for a while I get what I need.) Is it possible to somehow pay attention to this moment and record the situation?

PE5PVB, 2023-11-23:
> By the way, since the topic has started, I’ll say about the connection between the situation and the PS line :) Previously, I wrote everything about PS errors with a weak signal, but recently I [...]

This is caused by something else.

closed 2023-11-23 by PE5PVB