PE5PVB/TEF6686_ESP32 #456: Can't save frequency 87.5 FM to memory

Yurij20, 2023-11-29:
Can't save frequency 87.5 FM to memory.  Instead of saving, the memory cell is hidden and its number is shown in green

ohmytime, 2023-11-30:
This is the default channel value, I will change it to other values

ohmytime, 2023-11-30:
I just fixed this issue #458 ,please test.

PE5PVB, 2023-11-30:
This whole issue will be fixed in a future release when we add a skip function for memory channels.

ohmytime, 2023-12-01:
> This whole issue will be fixed in a future release when we add a skip function for memory channels.

Now , my solution can't save FM 108.00 to channel (smile)

Overland-DX, 2023-12-01:
Can't you just change the value to an invalid value? 
Since empty memory slots are not tunable anymore anyway?

#define EE_PRESETS_FREQUENCY 10800  --> #define EE_PRESETS_FREQUENCY 00000

I think I've tested that value before, but can test again when I get the Arduino IDE installed again.

Overland-DX, 2023-12-05:
@ohmytime 
I have tested the value 00000 for a few days without finding any bugs. But there is probably a better solution.

PE5PVB, 2023-12-05:
Where did you change that value?

andimik, 2023-12-05:
EE_PRESETS_FREQUENCY, see above

Overland-DX, 2023-12-05:
> Where did you change that value?

In constants.h, Line 194: #define EE_PRESETS_FREQUENCY        00000

Yurij20, 2023-12-15:
Fixed. Thanks 😊

closed 2023-12-15 by Yurij20