PE5PVB/TEF6686_ESP32 #462: Again AF method B detection

lxsxl, 2023-12-06:
Seems that the last development of method B detection has been eliminated. Obviously it works now as it did before the last enhancement (AF-B and one RV are again detected on Dlf 102,7).

PE5PVB, 2023-12-07:
Can you check if this is still the case when AF sorting is switched off?

lxsxl, 2023-12-07:
I was not able to reproduce it when AF autosorting is disabled. The head frequency (which is transmitted twice) is then listed only once in the first position.

Yurij20, 2023-12-16:
I confirm.  On video I recorded an error in the indication of the af-b method using the example of two stations 104.2 where the af-b method is available and 104.7 where it is not.  I fixed the problem after changing the station from ab-b to 104.7.  The af-b flag is set falsely despite the absence of the af-b method on 104.7

https://youtu.be/KQ0xI5JHNfI?si=6NEezaZ3Cw6Yq_kJ

Yurij20, 2023-12-30:
Hello, when will you look into this problem?  I would also like to add that the af-b method is not detected with a strong signal when rds reception occurs without errors

PE5PVB, 2023-12-30:
> Hello, when will you look into this problem? I would also like to add that the af-b method is not detected with a strong signal when rds reception occurs without errors

I don't understand what you mean with this.

lxsxl, 2023-12-30:
Often method B is detected immediately but sometimes - even if the quality of reception is without errors - it takes about 30 seconds (estimated).
But we still have wrong detection of method B in the special case that the broadcaster transmits the tuned frequency double (as head-AF and then again within the AF-list). It seems that the program checks again if there is at least one frequency more than once between two "number of AFs" (like it did until first or second RC). As far as I remember it was changed in RC2 or RC3 to "if the head-AF is transmitted at least in 50 % of the (head-)AF-list".
I will flash an older version to check if it definitely works different than the final releases.

PE5PVB, 2023-12-30:
You mean the issue which is there when AF sorting is enabled?

lxsxl, 2023-12-30:
Yes, obviously it is only in case that sorting is enabled. Whatever may be the reason for this...

PE5PVB, 2023-12-30:
I'll fix that issue.

lxsxl, 2023-12-30:
Okay, thanks!
Btw: I just flashed RC3, and indeed, method B detection is better than in V2.00. Probably an error during code cleanup?

closed 2024-01-27 by PE5PVB