PE5PVB/TEF6686_ESP32 #468: Automatic activation of underscores

Mike86-CZE, 2023-12-28:
If I have the option "Show RDS errors" set to "All" and I tune in to a frequency with a high error rate, after a while the option "Show underscore in RDS" is activated by itself, which can no longer be deactivated in the settings.

https://github.com/PE5PVB/TEF6686_ESP32/assets/140615384/ac87b9fd-90e1-4ae7-9627-944e0f75966a

PE5PVB, 2023-12-29:
I have no idea what causes this, I can't reproduce it and I don't see any way in the firmware which can cause this.

Mike86-CZE, 2023-12-29:
Note that the error only occurs when almost all the dots are red.  In addition to the underscores, the "Show PI errors" function is also activated, which cannot be deactivated either.

https://github.com/PE5PVB/TEF6686_ESP32/assets/140615384/9ee09d95-a1be-475b-a2de-03e051401594

Mike86-CZE, 2023-12-29:
I am updating the list of all RDS settings, which are then blocked and cannot be changed.
Show underscore in RDS
RDS filter
Show PI errors
RT buffering
Auto sort AF
Fast PS

PE5PVB, 2023-12-29:
It looks like a glitch in the memory somewhere. Will be very difficult to find.

PE5PVB, 2024-02-14:
I moved the variable in the code to another struct. It seems to be caused by a memory leak. Since the software is that complex I have no way of finding it. Maybe this did the trick.

Mike86-CZE, 2024-02-24:
The issue was partially resolved in version 2.00.6.  The underscores are no longer activated, but the "RDS filter", "Show PI errors", "RT buffering", "Auto sort AF" and "Fast PS" settings are still frozen (cannot be changed).

PE5PVB, 2024-02-27:
Can someone confirm this?

andimik, 2024-02-28:
I personally see no reason why someone should use "Show RDS errors = all" (as this generates too much issues), so I will not test that, sorry.

PE5PVB, 2024-03-04:
Yes, I agree with you. Setting to showing all errors just doesn't make sense.

closed 2024-04-01 by Mike86-CZE