PE5PVB/TEF6686_ESP32 #486: Overlaying the beginning and end of the text in the RT line after entering and exiting the analyzer

Yurij20, 2024-01-28:
After today's update, when entering and exiting the analyzer, the beginning and end texts are superimposed in the RT line

https://youtu.be/ZfxdAT-DKhA?si=0kHyuxsBOI_xidHc

PE5PVB, 2024-02-02:
Looks like this is caused by a problem in the library. I contacted the author.

Yurij20, 2024-02-12:
I tested the update all day.  Everything is fine, I didn’t notice any more text overlays.  Thank you

closed 2024-02-12 by PE5PVB

Mike86-CZE, 2024-02-17:
https://github.com/PE5PVB/TEF6686_ESP32/assets/140615384/387c8831-3e33-4b4b-829b-b54d654b7aca

I updated to 2.00.5 today and the text is overlapping for a single station.  For other stations, the text does not overlap, but the length of the space varies.  For some stations the gap is a few characters long, for others it is almost the entire line of the display.