PE5PVB/TEF6686_ESP32 #494: Enlarge the RT line by a few pixels lines

andimik, 2024-02-12:
Please note that some characters like German `Ö` are too large for the RT line.

In the example you will see every dots on the `Ö` where they have appeared first, and they will not be integrated in the ticker.

You can use the very few pixel lines from the AF line, as there you will **never** have those large characters.

[VID_20240212_215849.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/6e880b12-02e0-4729-a74a-f853daaff41d)

PE5PVB, 2024-02-12:
Well the text is wrapped. If I make the sprite less large some characters will miss the bottom. I'll investigate if I can find a solution somehow.

andimik, 2024-02-12:
No, the font should stay the same. Move the blue line and the RT up by 2 pixels and try.

PE5PVB, 2024-02-12:
That is not the solution. The space you want to remove is for some special chars in use. If I move it up 2 pixels, the dots on the ë are gone.

PE5PVB, 2024-02-17:
We updated the font. Can you please check if this solved your problem?

andimik, 2024-02-19:
yes, this solved it indeed.

andimik, 2024-03-15:
Again the case in d96a2e576adf135f9025944ad332841494d2444f

The remaining sprites are grey, not yellow.

PE5PVB, 2024-03-26:
What do you mean? You refer to a change in the language file.

andimik, 2024-03-26:
This is the software commit I was using for the issue, it contains all previous ones of course. 

As I am not a programmer I can't find out which exact commit causes the problem.

PE5PVB, 2024-04-01:
Please confirm if the problem is fixed. If not, post another screenshot.

andimik, 2024-04-02:
I'm still waiting for a fix ...

Have tested 44bab28043ae87ef17fc1ac3e7a74b9b39927f21 from today.

You can still see the dots from the letter `Ö` but at the bottom.

[VID_20240402_195225.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/b30c0344-504b-47c6-a4c3-9befab13a77f)

[image]
[image]

andimik, 2024-04-02:
Here another example with `Pyotr Tchaikovsky` alias `Peter Iljič Čajkovski`

https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/3c794a2a-c8ae-4636-9330-42722a079c07

[image]

PE5PVB, 2024-04-05:
Fixed.

andimik, 2024-04-06:
No, not fixed. Dots still remain there when `Ö` is received.

[image]

PE5PVB, 2024-04-06:
Can you give me a full RT line where I can test with?

andimik, 2024-04-06:
I could record an MPX file in the next days

PE5PVB, 2024-04-06:
No, just a line of text is ok. I can hardcode this in the software to reproduce the problem.

PE5PVB, 2024-04-06:
[image]

Are you sure you compiled the latest repository?

andimik, 2024-04-06:
Yes, with git pull

andimik, 2024-04-07:
I checked again, it was **after** the font update.

```
git show
commit eb5c1d600316d794834783d1f0e6f67a9cc9c1d1 (HEAD -> main, origin/main, origin/HEAD)
Author: Sjef Verhoeven PE5PVB <pe5pvb@het-bar.net>
Date:   Fri Apr 5 21:35:03 2024 +0200

    Font update
```

PE5PVB, 2024-04-07:
Then please give me a full textline to test with.

andimik, 2024-04-07:
Seems you just need RT text examples.

See http://andimik.bplaced.net/rds/spyfiles/A201_-_2023-12-01_19-28-08.txt

```
RT History:
19:25:48 Jetzt in Ö1: Öffentlich-rechtlich auf Bewährung                 
19:26:09 Für den ORF markiert der Jahreswechsel auch eine Zeitenwende.   
19:26:39 Die Finanzierung des öffentlich-rechtlichen Rundfunks ist ...   
19:27:06 E-Mail an Ö1: oe1.service@orf.at                                
19:27:39 Nächste Sendung: Víkingur Ólafsson und seine Goldbergvariationen
```

But I don't have made an mpx recording yet.

PE5PVB, 2024-04-08:
Problem is the font in combination with the library. I was able to reproduce it with your texts. We are now looking into a solution.

PE5PVB, 2024-04-10:
I found a solution by modding the library. This will be included with the next beta.

PE5PVB, 2024-04-29:
Please confirm if the problem has been solved.

andimik, 2024-04-30:
No, still not fixed in 2.00.15

PE5PVB, 2024-05-05:
I tested it with your sample line. Jetzt in Ö1: Öffentlich-rechtlich auf Bewährung  
I don't see any ghosting anymore. Please note, you need to update the modified tft_eSPI library if you compile it yourself.

andimik, 2024-05-17:
Are you talking about https://github.com/ohmytime/TFT_eSPI_DynamicSpeed ? 

This did not work, now my screen is white, but I can hear sound.

PE5PVB, 2024-05-17:
Then check the configuration file of tft_espi

PE5PVB, 2024-05-17:
Or just use a recent development build

andimik, 2024-05-17:
you're too fast, I think I found the error in Arduino IDE.

Let me try now and then I guess I can close the issue.

andimik, 2024-05-17:
> Or just use a recent development build

I went to discord

> v2.10 RC2
> 
>    Added menu option to configure 5s band button press
    Optimised GUI view SPI speed selector
    Fix problem with memory tuning when band is not FM
    Added support for RDS Long PS
    Fixed linebreak (replace with space) in RDS parser
    Font Update
    Language updates

but it still shows these dots. I did even a reset.

Therefore, the issue **cannot** be closed.

PE5PVB, 2024-05-17:
What is the exact text so I can reproduce?

andimik, 2024-05-17:
Any text with larger capital letters. I have that on different radio stations, Slovene `ČŽŠ` and German `ÄÖÜ`

Just use `Österreich` or `Ö1` and you will see that.

[VID_20240517_195433.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/cf90ce85-0156-4b77-9d08-12824c6ea8d3)

PE5PVB, 2024-05-17:
OK. Give me a sampletext I can play with.

andimik, 2024-05-17:
Do you need an mpx file?

PE5PVB, 2024-05-17:
No just a RT text as text. I can use this text for debugging.

andimik, 2024-05-17:
Use `Österreich Österreich Österreich Österreich`, this should be enough. It seems that the ticker needs to be scrolling the text.

PE5PVB, 2024-05-17:
OK. Please check RC3, which I just posted.

andimik, 2024-05-17:
Just tested RC3, slightly better, but the dots are **still there**, they are just a bit brighter (difficult to see on the video) 

`GEORG PHILIPP TELEMANN - HAMBURŠKA GLASBA NA VODI TWV 55:C3 (V Ž`

You can test it with that mpx file

https://we.tl/t-mnXwRflnCX

[VID_20240517_212849.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/36c34900-26fc-4a76-9dc3-0e8949bae30c)

PE5PVB, 2024-05-17:
OK. I guess you are able to compile your own build. In the libraries\TFT_eSPI\Extensions folder there is a file called Smooth_font.cpp. On line 133 you will find:

`gFont.descent  = (uint16_t)readInt32() + 2; // bottom of "p"`

Try to change the + 2 to a higher value, start with 3 and find the spot where the problem is gone.

andimik, 2024-05-17:
Ok, I'll try, but first I need to find out why the screen gets white when I am uploading the compiled software. This only exists since I've updated this library.

PE5PVB, 2024-05-17:
Can you send me a PM via Discord, so I can prepare a testversion for you?

andimik, 2024-05-18:
Changed the value in
```
gFont.descent  = (uint16_t)readInt32() + 2; // bottom of "p"
```
to `4` and problem has gone.

[VID_20240518_093625.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/ebb5418c-81cc-47dc-bdd3-7582b277810a)

[VID_20240518_093908.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/d21a58ce-89f8-4523-a537-cbd38939873f)

lxsxl, 2024-05-25:
Foud glitches again; this can be seen with german language setting in AF and in EON mode:
[image]

PE5PVB, 2024-05-26:
Just checked it, but with the current build I'm working on I can't reproduce this anymore.

andimik, 2024-05-26:
Confirmed, it's gone.

 @lxsxl please install the discord version. Seems you were doing the same mistake and compile it from source with outdated eSPI library.

Therefore, I close it.

closed 2024-05-26 by andimik

lxsxl, 2024-05-26:
Okay, with the new library it has disappeared.