PE5PVB/TEF6686_ESP32 #498: Another problem with the segment font

leryfm, 2024-02-18:
Hello Sjef!
 
Why the segment font has been changed again in the latest firmware 2.00.6?
I suspect that the goal was to make it closer to what it was before August 2023, but believe me, the one from January 2024 is much better. 

The numbers are more correct in shape and more beautiful. Please return it. For what reason can you not make two variants? "Classic" / "Classic 2", both of them are on the list if you found the opportunity to return the old one? (at the same time, it still does not have 100% similarity with the one that was)
To be honest, I felt disappointed again, just like when the segment font disappeared completely. Moreover, the rest of the fonts were not affected; this is clearly not a return to the old library or its replacement with another new one. I’m not insisting on anything, but I would like to know why this particular font needed to be changed again, because this is an important point that I asked about earlier... 

Thank you for any answer. I hope that two of these fonts can be saved, if someone needs a second option that replaced the previous one.
But I really wouldn’t like to stay on the old firmware... This is a very important point

alexstarvrn, 2024-02-18:
I support it, the font that was in the previous firmware in January looked better. Please return it.

Yurij20, 2024-02-18:
I support it, the Classic font has unfortunately become worse in appearance, some segments have begun to look uneven

PE5PVB, 2024-02-27:
Font has been changed for the last time.

Yurij20, 2024-03-01:
I checked the classic font and it became much better, but the middle segment in numbers is still too long and doesn’t look very good.  If it is possible to shorten this middle segment and sharpen the lateral vertical segments of the central one in the center?) In the photo I carefully showed how) Thank you).  In the Motoya font, if you look closely, the numbers in height have become uneven, if you look closely, it’s insignificant, but it’s there.  At 2.00.6 everything was perfect.  I would return it as in 2.00.6
[image]

PE5PVB, 2024-03-01:
No photo...

Yurij20, 2024-03-01:
[image]

Yurij20, 2024-03-01:
https://disk.yandex.ru/i/cO6F81ktYzvVAw

andimik, 2024-03-01:
@Yurij20

You need to wait till the upload was successfully and then submit.

Yurij20, 2024-03-01:
[image]

leryfm, 2024-03-04:
Dear Sjef! 

Please return the font that was presented in the firmware at the beginning of January 2024. 
It was the most beautiful, better than all the previous ones. It was a real New Year's gift for me, very nice, and now it has disappeared again. 
I don't understand why it needed to be changed and removed again. 
Perhaps it was necessary to replace the library again, but all other fonts added with the new library were not touched. 
Why can’t you **leave both versions of the segment font** if you don’t like the one in the photo and want to change it? I kindly ask you to return the font and not remove or modify it.

[image]

PE5PVB, 2024-03-04:
We will do some small adjustments to the classic font and added a modern 8-digit font extra. And there we draw a line. If you want to do more adjustments, you are welcome to do it yourself.

leryfm, 2024-03-04:
Thank you for your reply. Did I understand you correctly that the font with which I attached a photo above will be returned? Or are we talking about something else called "modern 8-digit font extra"?

I would like to clarify: Just bring back that segmented font that was in January '2024. Exactly that one, without changing or correcting anything. It doesn't need fixing, smoothing etc. Just like it was in beta 2.00.3 and so on, and then for unknown reason it disappeared. I'm talking about the font in the photo I sent above. It's great and I even wrote you a happy review...

It’s a strange decision to remove it after that without explanation....
“Do it yourself” is an easy answer, but I don’t understand why this font doesn’t deserve to be present in the main branch of the firmware along with the others, especially if it was already there, and it's the best!

zygmund2000, 2024-03-08:
Adjusting anti-aliased fonts is very difficult, the processing tool does it very poorly. For me it is pain for eyes.
If anyone knows an alternative method to generate it, please let me know

closed 2024-04-05 by PE5PVB