PE5PVB/TEF6686_ESP32 #505: Unwanted interference on some frequencies on the medium wave when the screen is off.

Mike86-CZE, 2024-03-09:
In the last few versions of fw there is a new interference on medium wave that comes directly from the receiver.  The interference is only on some frequencies and only occurs when the screen is off.  Interestingly, pressing the button has an effect on interference.

https://github.com/PE5PVB/TEF6686_ESP32/assets/140615384/7ffec651-7feb-423a-91c9-0a69c2e68755

PE5PVB, 2024-03-09:
I tested this with my radio, but was not able to reproduce. I think it's something specific in the hardware which causes this. Can't be fixed by software.

Mike86-CZE, 2024-03-09:
I know for sure that this did not happen in earlier versions of fw.  Something must have changed in fw recently to cause this.

PE5PVB, 2024-03-09:
Let me know at which version the problem is introduced.

Mike86-CZE, 2024-03-10:
It seems that the screen off is not necessarily a requirement.  Note that in this case I have the antenna completely disconnected.  I will try to find out in which version the interference appeared.

https://github.com/PE5PVB/TEF6686_ESP32/assets/140615384/a038fd4d-8255-4824-a172-1ac61a832e5b

Mike86-CZE, 2024-03-10:
2.00.1 no interference
2.00.2 no interference
2.00.3 no interference
2.00.4 is interference
2.00.5 is interference
2.00.6 is interference
2.00.7 even more interference
tested without antenna, with screen off on frequencies
1269, 1305, 1350, 1386, 1440, 1467, 1539, 1548, 1584 kHz.
 I also found that the Squelch on/off option has an effect on interference.

PE5PVB, 2024-03-10:
If you switch off squelch, problem is gone?

Mike86-CZE, 2024-03-10:
no, there is interference in both cases, but it affects the intensity of the interference, the effect of pressing the buttons, the sound of the noise, etc.

PE5PVB, 2024-03-10:
Are you able to compile code yourself?

Mike86-CZE, 2024-03-10:
no, unfortunately I can't.

PE5PVB, 2024-03-10:
OK. I'll look into it when I have the time for it. Thanks for your report.

PE5PVB, 2024-03-12:
First thing what comes in mind is the increase of SPI speed. I'll set it back to previous value with the next development beta.

Mike86-CZE, 2024-03-16:
It seems to have helped.  Interference has significantly decreased, or rather it is approximately in the previous state.

closed 2024-04-01 by Mike86-CZE