PE5PVB/TEF6686_ESP32 #511: switch encoder standard/optical not working correctly

Mike86-CZE, 2024-03-16:
After updating to 2.00.9 (I skipped 2.008) the encoder settings have no effect.  In both cases, when rotated, the frequency sometimes jumps two positions (e.g. 200 kHz instead of 100 kHz)

PE5PVB, 2024-03-16:
Was this not the case in previous versions?

Mike86-CZE, 2024-03-16:
in version 2.00.7 and earlier I did not encounter this problem. It was happened with the "optical" setting, but it was fine with the "standard" setting.

PE5PVB, 2024-03-16:
OK. I did replace the version about half an hour after releasing it via Discord. Can you try to update it again to make sure if the problem is still there, because I changed something there.

Mike86-CZE, 2024-03-16:
I downloaded it today.  was inserted on 14.3. 2024 20:10.  So I should really have the current version.

PE5PVB, 2024-03-16:
Ah ok. I will fully revert this change because of debouncing of some cheap rotary encoders.

closed 2024-04-01 by Mike86-CZE