PE5PVB/TEF6686_ESP32 #519: A very fast frequency change will cause a restart

Mike86-CZE, 2024-04-01:
If I control the TEF6686 via WiFi and change the tuned frequency very quickly, the receiver restarts.  Tested so far with Stationlist-M.  Can someone verify in XDR-GTK?  v2.00.10

PE5PVB, 2024-04-01:
I'll look into this, is this only on AM or also on FM?

Mike86-CZE, 2024-04-02:
both.  FM and AM.

andimik, 2024-04-02:
I cannot reproduce it with xdr-gtk's spectrum when I scroll from one end to the other within less than one second.

[VID_20240402_202321.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/1128fc78-89e6-45f6-a9c1-295764d6c1db)

Mike86-CZE, 2024-04-02:
Try switching to AUTO or MEM and repeat.

Mike86-CZE, 2024-04-02:
According to the battery voltage indicator, wifi is not active, but it is connected via USB.  Try over WiFi.

andimik, 2024-04-03:
I don't know why, but I cannot configure WiFi with 2.0.11. ESP_xxx is full signal, but I cannot connect.

So I have tested xdr-gtk with AUTO and MEM but had no problems with very fast frequency changes.

Could you test, if you can configure WiFi?

PE5PVB, 2024-04-03:
Found a memoryleak which could cause this. Try recompile the current branch and let me know.

andimik, 2024-04-03:
Even without the recent memory leak fix I could not reproduce it with Stationlist-M and WiFi.

Mike86-CZE, 2024-04-03:
It took me quite a bit of work to trigger the error today.  Yesterday it happened much more often.  I have no idea why.

https://github.com/PE5PVB/TEF6686_ESP32/assets/140615384/d751570b-0411-4230-a269-ae1a975008d9

andimik, 2024-04-03:
According to the video please improve the WiFi signal. I fear this is causing the issue.

Mike86-CZE, 2024-04-03:
Yes, I'm pretty sure it affects the quality of the wifi connection.  I have now placed the TEF further away from the router (same location as yesterday) and managed to trigger the error three times in a short while.  However, earlier versions of fw did not have this problem.

Mike86-CZE, 2024-04-03:
> According to the video please improve the WiFi signal. I fear this is causing the issue.

We seem to have the same TEF model from the same manufacturer.  Did you modify the wifi antenna somehow for better range?  My TEF has very poor wifi range, only a few meters and preferably with line of sight.

PE5PVB, 2024-04-03:
Can you monitor the data on the USB port with a terminal program like Putty? It should show you a message at those reboots.

andimik, 2024-04-03:
A better Wi-Fi is not necessary. I am placing the mobile next to the the TEF.

PE5PVB, 2024-04-03:
Very curious about the serial output when the reboot happens....

Mike86-CZE, 2024-04-06:
I have found that if the screen saver is activated (display is off) then reboots do not occur.  Once I activate the screen again, it's easy to trigger the error/restart.

PE5PVB, 2024-04-14:
Please check again with v2.00.13b

Mike86-CZE, 2024-04-15:
So far I have not been able to trigger a reboot.  The bug really seems to be gone.  Good work.

PE5PVB, 2024-04-29:
OK. Let's close this one for now.