PE5PVB/TEF6686_ESP32 #520: Return of positive squelch result is missing the newline terminator

clorenz, 2024-04-01:
In `comms.cpp`, line 417, the newline is returned too early, and the value is sent afterwards without final newline.

Suggestion:
```
          DataPrint("Q");
          DataPrint(String(Squelch));
          DataPrint("\n");
          Squelch *= 10;
```

PE5PVB, 2024-04-01:
Thanks. Changed it.

clorenz, 2024-04-02:
Thank you very much.

Unfortunately, the latest version (Branch `main` as of Apr 2nd) results in a full white screen after being upload via Arduino. Had to revert back to latest discord version 2.0.11 w/o the fix.

Am I missing something?

PE5PVB, 2024-04-02:
I just checked the branch, but have no problems uploading it. Please check your tft_espi configuration.

clorenz, 2024-04-02:
I've found it - I did only enable the required defines, but did not disable all other defines.

Thank you very much for your help :-)

Next is to figure out (but not today anymore), why the WIFI setup does not work, my mobile does not connect to the TEF WLAN any more. With 2.0.10, it was possible.

PE5PVB, 2024-04-03:
Found a memoryleak which could cause this. Try recompile the current branch and let me know.

clorenz, 2024-04-03:
Work perfectly now. Thanks a lot!

closed 2024-04-03 by clorenz

blazer221, 2024-04-03:
With the latest commit that fixes the leak the wifi AP works fine for me (it didn't refresh the page on the TEF for some reason, also i think it will be a good idea to have an option to go back from that page with AP info). Another issue that's present for some time is that when Wifi is active (the wifi meter shows some singnal) the battery voltage is not present. Just an empty battery.

PE5PVB, 2024-04-03:
Correct. I can't cancel the WiFi setup, this is how the WiFi library is build. Only way to cancel it, is to powercycle the radio.

And it's correct that your battery indicator is not working when WiFi is connected. This is a hardware limitation on the ESP32.

PE5PVB, 2024-04-03:
Yup.