PE5PVB/TEF6686_ESP32 #526: RDS Block D indicator (dot) is blinking

andimik, 2024-04-09:
Although the signal is error-free, the RDS Block D quality indicator is blinking.

I have the problem on all frequencies, even local ones, so I fear there is a bug in the calculation.

[VID_20240409_065559.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/490b77ce-e412-4d8c-a74e-79f584a2adb0)

andimik, 2024-04-09:
Updated: Then the output is not error-free:

```
{"pi":"0xA203","group":"8A","bler":13,"prog_type":"Pop music","tmc":{"message":{"description":"Contraflow. Construction work.","direction":"single","event_codes":[517,803],"extent":"-0","location":35926,"update_class":5,"urgency":"none"}},"tp":true}
{"pi":"0xA203","group":"0A","bler":13,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"14A","bler":6,"other_network":{"pi":"0xA902","kilohertz":95400,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","bler":13,"di":{"compressed":false},"is_music":true,"ta":false}
{"pi":"0xA203","group":"0A","alt_frequencies_b":{"tuned_frequency":99900,"same_programme":[88800]},"bler":13,"di":{"compressed":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"2A","bler":13,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"8A","bler":13,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":13,"di":{"artificial_head":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"2A","bler":6,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","bler":13}
{"pi":"0xA203","group":"14A","bler":13,"other_network":{"pi":"0xA902","kilohertz":98100,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"8A","bler":13,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","ps":"  OE 3  ","alt_frequencies_b":{"tuned_frequency":98600,"same_programme":[99500]},"bler":13,"di":{"stereo":true},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"2A","bler":13,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"14A","bler":13,"other_network":{"pi":"0xA902","kilohertz":94600,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":6,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","bler":13}
{"pi":"0xA203","group":"8A","bler":13,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":13,"di":{"compressed":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"14A","bler":13,"other_network":{"pi":"0xA902","kilohertz":94100,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":13,"di":{"artificial_head":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"2A","bler":6,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"8A","bler":6,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","bler":13,"di":{"stereo":true},"is_music":true,"ta":false}
{"pi":"0xA203","group":"0A","ps":"  OE 3  ","bler":13,"di":{"stereo":true},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"3A","bler":13,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"Pop music","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":true}
{"pi":"0xA203","group":"14A","bler":13,"other_network":{"pi":"0xA902","kilohertz":93200,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":13,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"8A","bler":6,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":6,"di":{"compressed":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","bler":13}
{"pi":"0xA203","group":"14A","bler":13,"other_network":{"pi":"0xA902","kilohertz":96800,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":13,"di":{"artificial_head":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"2A","bler":13,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"8A","bler":13,"prog_type":"Pop music","tmc":{"message":{"description":"Closed.","direction":"both","event_codes":[401],"extent":"+3","location":42785,"update_class":5,"urgency":"U"}},"tp":true}
{"pi":"0xA203","group":"4A","bler":6,"clock_time":"2024-04-09T07:20:00+02:00","prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","ps":"  OE 3  ","bler":6,"di":{"stereo":true},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","bler":13}
{"pi":"0xA203","group":"2A","bler":13,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"14A","bler":13,"other_network":{"pi":"0xA902","kilohertz":95700,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"8A","bler":13,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":13,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"2A","bler":6,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","bler":13}
{"pi":"0xA203","group":"14A","bler":13,"other_network":{"pi":"0xA902","prog_type":"No PTY","ta":true,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":13,"di":{"compressed":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"8A","bler":13,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","alt_frequencies_b":{"tuned_frequency":98300,"same_programme":[90400,99500]},"bler":13,"di":{"artificial_head":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
[... cut ...]
{"pi":"0xA203","group":"3A","bler":13,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"Pop music","tmc":{"system_info":{"gap":3,"service_id":0}},"tp":true}
{"pi":"0xA203","group":"14A","bler":6,"other_network":{"pi":"0xA201","tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":6,"di":{"compressed":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"8A","bler":6,"prog_type":"Pop music","tp":true}
```

BER of 13 seems to be 12.5%
BER of 6 seems to be 6.25%

PE5PVB, 2024-04-10:
I tested this with 2 radio's and not able to reproduce. Can you check if others experience the same problem? Maybe I can set the error level indicators just a little less sensitive, because they now react on every small error, even when repairable.

andimik, 2024-04-10:
I fear that the culprit is the option "RDS filter". Default after a reset or new flash is "Off". Rdsspy has 20% BER :-(. 

If you set it to "On", the blinking will disappear and I have 0% BER in Rdsspy.

Can you explain that and make "On" as default?

PE5PVB, 2024-04-11:
When switched off the raw unfiltered RDS blocks are processed. This will increase the RDS decoding, but can also give you some errors. When enabled, RDS blocks are rebuild by the TEF-chip. Processing will be much smoother, but has some small lag and is somewhat slower.

andimik, 2024-04-11:
https://github.com/PE5PVB/TEF6686_ESP32/commit/303c193033b3552e7d990136bf87b2a795daa9f9#diff-E6C79ED4678FAE99BD53671CD5A1CCC0

I will test it soon. But I still don't understand, why only Block D has the problem, and A, B, C not.

PE5PVB, 2024-04-11:
Well, it's just a raw and ugly mode. Propably the used RDS encoder just works somewhat different what triggers your problem.

andimik, 2024-04-11:
So you mean, it could be hardware related?

PE5PVB, 2024-04-11:
No, I'm not talking about your radio, but about how the RDS encoder at the transmitter site works.

andimik, 2024-04-11:
Not agreed. My problem occurs on stations from all 3 countries I can receive (Austria, Italy and Slovenia). So it has nothing to do with the transmitter.

PE5PVB, 2024-04-11:
OK. Well, I'm not able to reproduce it, but again, with buffering off, it's a raw decoding, so it is possible these things can be triggered by unknown causes. When using external RDS decoding software like RDS Spy, it's best to enable the buffering to make sure you send the correct blocks to the decoder.

andimik, 2024-04-11:
I will test all my TEFs in the next days with latest (and previous) commit.

andimik, 2024-04-11:
No, this did not help on two different TEFs, still blinking in RDS group D and higher BER in RDSSpy/redsea. So it's not hardware related and seems to be a bug in the firmware.

All frequencies affected.

Will test another 2 TEFs, but I don't think they will behave different.

andimik, 2024-04-11:
Flashing with the Discord Version did not help. It has nothing to do with Ubuntu and Arduino IDE.

I wonder that nobody has complained that before. On a 3rd TEF I have 2.00.4, where it is blinking as well.

So it's on all my TEFs.

Hint: it does not matter if I have WiFi on or off, if a USB connection is established or not.

PE5PVB, 2024-04-11:
RDS decoding shouldn't have any relation with WiFi or USB connection. So my advice for you is to enable RDS buffering, because you use an external RDS decoder.

andimik, 2024-04-12:
I found out, that when I activate the screensaver and switch the display off, I have 0% BER on all frequencies.

So it has to do with the display!

andimik, 2024-04-12:
Update, it's not exactly 0%, look at the interesting pattern (captured with radio signal from AUT, SLO and ITA)

[image]
[image]
[image]
[image]

andimik, 2024-04-12:
I guess I found the bug.

I have recorded the same radio station with a) TEF and b) SDR Stick and compared the output.

The pattern shown above in the RDSSpy screenshots is a result of duplicate RDS Block A, B and C messages every **around** 150th RDS line (exact numbers see below). **This explains why Block D is blinking.**

See screenshot, left is SDR stick, right is TEF. Have a look at the pattern (the diff) on the very right end. Signal was errorfree in both cases. But TEF adds a "ghost" RDS Block A, B and C (Block D is skipped), which is not transmitted.

[image]

[tef_versus_rtlstick.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/14963478/tef_versus_rtlstick.zip)

If you want to know the offset, it's not exact every 150th line, but it's

139
161
143
155
147
157
148
148
149
154

andimik, 2024-04-12:
And this is the comparison with "RDS filter = on". No errors, both show the same content (the green bar is the start and end of the longer file on the left).

Please set this option on as default.

[image]

andimik, 2024-04-17:
> Can you check if others experience the same problem?

Confirmed by three other users on Github.

PE5PVB, 2024-04-17:
I think I found the cause. Please test.

andimik, 2024-04-17:
You mean the bugfix some minutes ago on AF, BF, CF and DF instead of A, B, C and D?

PE5PVB, 2024-04-17:
No this commit: https://github.com/PE5PVB/TEF6686_ESP32/commit/a854966f193794bddb44efd0f2e75e0e57d0f5fb
The other one was just for testing.
I'll draft a beta in an hour.

PE5PVB, 2024-04-17:
Please check the compiled version from Discord. Also confirm if the blinking of block D is fixed.
If you set RDS filter ON all is OK?

andimik, 2024-04-17:
Well, this was the same version, no matter, if github or discord of course. But I have installed it anyway (incl. formatting process) as you asked to do so.

# RDS filter = on

no blinking in Block D, no errors, no fake blocks

But slower display on TEF, PS will take some seconds till displayed.

[image]

# RDS filter = off

blinking in Block D, BER is not 0%, fake RDS blocks

[image]

It seems that the problems are worst right after tuning to a frequency. The longer you stay on the frequency, the better the results.

Here another result on the console:

# RDS filter = off

```
$ nc 127.0.0.1 7376 | redsea -hx | grep -E '[[:punct:]]' -A 1
A540 054F 88AE ----
A540 054F 88AE 4520
--
A540 2545 6C79 ----
A540 2545 6C79 2046
A540 0549 8892 ----
A540 0549 8892 5445
A540 054A 88A8 ----
A540 054A 88A8 4E4E
--
A540 054F ---- ----
A540 054F 5288 4520
A540 0548 ---- ----
A540 0548 ED52 414E
A540 2540 4C69 ----
A540 2540 4C69 6E6B
A540 0549 5256 ----
A540 0549 5256 5445
--
A540 2541 ---- ----
A540 2541 696E 2050
A540 054F ---- ----
A540 054F 52AE 4520
A540 0548 ---- ----
A540 0548 52C7 414E
A540 2542 6172 ----
A540 2542 6172 6B20
A540 0549 5292 ----
A540 0549 5292 5445
--
A540 2543 ---- ----
A540 2543 2D20 4672
A540 054F ---- ----
A540 054F EDAE 4520
A540 0548 ---- ----
A540 0548 AEC7 414E
--
A540 0549 ---- ----
A540 0549 A8AE 5445
A540 054A ---- ----
A540 054A 92AE 4E4E
A540 2545 ---- ----
A540 2545 6C79 2046
A540 054F ---- ----
A540 054F 88AE 4520
A540 0548 ---- ----
A540 0548 56AE 414E
A540 2546 ---- ----
A540 2546 6972 6520
A540 0549 ---- ----
A540 0549 52AE 5445
A540 054A ---- ----
A540 054A EDC7 4E4E
A540 2540 ---- ----
A540 2540 4C69 6E6B
A540 054F ---- ----
A540 054F AEC7 4520
A540 0548 ---- ----
A540 0548 A8C7 414E
A540 2541 ---- ----
A540 2541 696E 2050
A540 0549 ---- ----
A540 0549 92C7 5445
A540 054A 88C7 ----
A540 054A 88C7 4E4E
A540 2542 6172 ----
A540 2542 6172 6B20
A540 054F 56C7 ----
A540 054F 56C7 4520
A540 0548 52C7 ----
A540 0548 52C7 414E
A540 2543 2D20 ----
A540 2543 2D20 4672
A540 0549 EDA8 ----
A540 0549 EDA8 5445
```

# RDS filter = on

```
$ nc 127.0.0.1 7376 | redsea -hx | grep -E '[[:punct:]]' -A 1
^C
```

meaning there are NO errors even with weak FM signals.

PE5PVB, 2024-04-17:
I just build another approche on error detection. Can you test that?

andimik, 2024-04-18:
Wow, this looks very good. First short tests: No blinking, no fake blocks with RDS filter off.

Will run a longer test, but I don't think I'll find errors. 

-> closing that issue.

Thanks, and have a nice vacation!

andimik, 2024-05-28:
Bug is back

PE5PVB, 2024-05-28:
No, actually it's a long lasting bug which is fixed. You need to switch on the RDS filter. RDS filter ON is now the same as RDS filter OFF in the past.

RDS filter OFF: Just parse raw RDS blocks, even if incomplete. Usefull for very short openings (MS) or heavily fading.
RDS filter ON: Only parse RDS when all 4 blocks are received. 

When filter is switched on, when tuning into a radiostation, first RAW RDS data is used, untill all PS bytes are received. Then the filter is automaticly turned on. This is the reason why you see the RDS error indicators blink just after tuning, but result is a much faster PS direct after tuning.

When filter is switched off, it will always stay off, even when PS bytes are received.

This is exact how I want it to be, but I forgot to mention in the updatelog.

closed 2024-06-02 by PE5PVB