PE5PVB/TEF6686_ESP32 #527: WIFI configuration broken

clorenz, 2024-04-12:
With the latest `main`, the WIFI configuration is broken again.

Connecting to http://192.168.4.1 waits forever.

(BTW - is there a chance to store the WIFI credentials in the EEPROM, so that they survive a Firmware update via Arduino?)

andimik, 2024-04-12:
It's the other way round. Even after a new flash or a reset the Wi-Fi configuration is still available.

clorenz, 2024-04-12:
That's interesting, for me it's indeed the other way round. After the flash of the 2.00.13, xdr-gtk could not connect any more because of an invalid password, since the password has been reset to the default password.

andimik, 2024-04-12:
I don't care about the rotary direction, I already know the steps what to do after each flash.

clorenz, 2024-04-12:
Same here, but the broken WIFI configuration is nasty ;)

When I look at the code, I can remember, that after the flash, the first reboot led to a second reboot (with a short white screen between those reboots). Is it possible, that the following piece of code was executed: https://github.com/PE5PVB/TEF6686_ESP32/blob/main/TEF6686_ESP32.ino#L671-L678 ?

andimik, 2024-04-13:
But indeed, yes it's broken.

In the past 2 hours (connected thru WiFi and xdr-gtk) I had several reboots and :exclamation:  resets. Yes, it's a **reset** (rotary direction is set to default and WiFi was off. With the first 3 attempts I just needed to activate Wifi. But after the 4th reboot I cannot enable or configure Wifi any more.

Luckily I was at home, but I don't want to have that when I am travelling. :angry: 

> Message Corrupt

[image]

PE5PVB, 2024-04-14:
Please check again with v2.00.13b

andimik, 2024-04-15:
for me,  v13b looks better now. At least the setup works as fast as in the 2.00 final, BUT there are still freezes when Wifi connected is established.

PE5PVB, 2024-04-15:
You mean while it is connecting? It's normal the radio is frozen while WiFi connection is being build up.

andimik, 2024-04-15:
No, during I am recording RDS data with RDS Spy. I can retune to another frequency (you will listen the new frequency) but the display is frozen.

It's not happening each time, but I have it sometimes.

PE5PVB, 2024-04-15:
OK. This happens when you tune with RDS Spy?

andimik, 2024-04-15:
No, I tune with TEF. RDSspy is just displaying the RDS.

PE5PVB, 2024-04-15:
OK. So TEF reacts slow.... Are the meters also updated slow? Can you make a short video?

andimik, 2024-04-15:
Here you are, tested on a local frequency on 92.8 MHz, see video, even when changing to 92.9 the display stays the same.

[VID_20240415_232049.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/84609486-f065-4798-9c53-077a1a83777c)

PE5PVB, 2024-04-15:
OK. But the trigger is when you use RDS Spy to tune? Do you know in which version this problem was introduced?

andimik, 2024-04-15:
No, for that freeze, RDS Spy initially was NOT tuning. I just tested it in the video, but this was later.

I cannot tell you exactly, but I tested several TEFs and all are affected.

PE5PVB, 2024-04-15:
OK. And what version introduced the problem?

PE5PVB, 2024-04-15:
And is this on WiFi or serial or both?

andimik, 2024-04-15:
Wifi

PE5PVB, 2024-04-29:
Is this problem still actual with the latest beta?

closed 2024-05-12 by PE5PVB