PE5PVB/TEF6686_ESP32 #529: Battery voltage meter problem

Mike86-CZE, 2024-04-13:
The battery voltage meter sometimes shows an incorrect value.  Like now, it was showing 3.6V.  I turned the TEF off and on again and it shows the correct 3.9V.  With the new version 2.00.13 this has happened to me several times already.

PE5PVB, 2024-04-14:
Please check again with v2.00.13b

Mike86-CZE, 2024-04-21:
Unfortunately, this is the second time this has happened to me, even with a newer version.

PE5PVB, 2024-04-29:
Please note that the voltage readout will always be a bit buggy, because it's implemented in the hardware the wrong way.

closed 2024-05-17 by PE5PVB

Mike86-CZE, 2024-06-02:
Something must have changed in the last versions.  Now fully charged battery shows as 64% / 3.75V.  Previously, the charge was displayed quite accurately.

PE5PVB, 2024-06-02:
Yes, I already undo this change.