PE5PVB/TEF6686_ESP32 #530: Latest Arduino ESP32 board update breaks compilation

clorenz, 2024-04-14:
After installing the latest ESP32 board update, compilation is no longer possible: "Text selection exceeds available space in board":

```
Sketch uses 3024581 bytes (230%) of program storage space. Maximum is 1310720 bytes.
Global variables use 63172 bytes (19%) of dynamic memory, leaving 264508 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board
```

Any ideas, how to either revert the update, or to fix the available space?

PE5PVB, 2024-04-14:
I guess you didn't set the compilation size to Hugh app in the compiler.

clorenz, 2024-04-14:
That's it :+1: The board upgrade reset it to a default.

closed 2024-04-14 by clorenz

zywek123, 2025-03-27:
I have similar problems.
arduino-cli compile --fqbn esp32:esp32:esp32 --build-property "build.partitions=huge_app" --board-options PartitionScheme=huge_app .
Sketch uses 3149721 bytes (100%) of program storage space. Maximum is 3145728 bytes.
Global variables use 63244 bytes (19%) of dynamic memory, leaving 264436 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.

How can i fix it? I have to use arduino-cli because arduino gui is not accessible for blind users on Linux.