PE5PVB/TEF6686_ESP32 #531: After reducing the sensitivity of the RDS error indicator, the PS line began to work better.

Yurij20, 2024-04-14:
After reducing the sensitivity of the RDS error indicator, the PS line began to work better.  I dealt with this problem once before.  We tried to solve it for a long time but it didn’t work out.  When you lowered the sensitivity of the error indicator, it became better, but not perfect yet.  I advise you to lower it even more

PE5PVB, 2024-04-29:
How does it respond with the latest beta?

PE5PVB, 2024-05-05:
Maybe disabling fast PS will help. Not much more I can do, otherwise we will loose compatibility with some other RDS encoders.

Yurij20, 2024-05-06:
disabling fast PS helps at startup, yes, but then everything goes the same way.  It’s a pity, of course, that the encoder for basic old firmware may give some incompatibilities, but PS works perfectly

PE5PVB, 2024-05-07:
Can you make a video of the effect you have and share a MPX recording?

PE5PVB, 2024-05-18:
@Yurij20 Any change of providing a MPX recording of the channel with problems?

Yurij20, 2024-05-26:
> @Yurij20 Any change of providing a MPX recording of the channel with problems?

Now I'm on vacation and while I'm away there's no way to do mpx.  When I arrive, I’ll do this) I still test the firmware on trips and 2.00 2.10, alas, the dynamic PS crumbles at the slightest weakening of the reception, unlike 1.15.  Now I’m using one of the modifications based on 1.15 where the PS encoder works as it should.  As for 2.00.15, I once again sharpened my attention to it and the improvements are insignificant, barely noticeable in my opinion.. Now you have added an improvement to the work PS I was chilling, when you make a new release I will check it and definitely tell you)

Yurij20, 2024-05-26:
I delved a little into the essence of the problem and realized that I had formulated it for you incorrectly, which is why we did not agree on an opinion earlier and could not solve it.  As for the errors that I hinted about earlier, everything has been fixed and when you disable them in the menu, they no longer appear.  The point is that when rotating the dynamic PS, at the slightest weakening of the reception, the next part moves from the previous one.  In the first firmware 1.15, etc., this happened but extremely rarely, and if the quality of RDS reception weakened, then the rotation of the dynamic PS was suspended until good reception quality was restored to avoid a similar situation.  Here, only when starting with the Fast ps mode disabled, the ps starts qualitatively without problems at any reception level, then the rotation occurs without reacting to the reception quality and therefore the rotation of the dynamic ps occurs with mixing.  I apologize that I may have formulated the essence of the problem incorrectly and I hope you will understand now and solve this situation.  An MPX recording was now made by a friend of mine at one of the stations where this problem occurs most often just in the right conditions where the firmware, for example, 1.15 rotates perfectly but not so fast, and 2.10 without slowdowns just with mixing.  I post it below:

https://disk.yandex.ru/d/8mhEvADRcsnNyQ

PE5PVB, 2024-05-26:
There are just so many PS packets missing in this file. I made an extra check to check if all 4 packets are received (each packet contains 2 bytes). And when the first packet is received just start counting again. This results in less errors, but also not all faulty packets will be shown. I Also have the idea that this encoder changes the PS in the middle of the packet transmission, in stead of from packet 0.

Let's see what this does.

Yurij20, 2024-05-26:
> There are just so many PS packets missing in this file. I made an extra check to check if all 4 packets are received (each packet contains 2 bytes). And when the first packet is received just start [...]

Yes, this station encodes a little differently PS noticed).  On a Samsung phone with a built-in FM radio, PS packages are read out of order, in firmware 2.00, 2.10 and even 1.15 everything is read in order) I'm waiting for the next update) I'll test it)

PE5PVB, 2024-05-26:
Well, if the radio misses packet 0 it can still show errors, but it's almost impossible to get it right with this signal.

Yurij20, 2024-05-26:
checked) at this station first.  Finally, it started working as it should, there are rare glitches despite the already absolutely terrible reception, but this was also in firmware 1.15, so everything is working fine) I’ll test it again tomorrow on a trip).  But another problem arose: the disabled Fast Ps mode stopped working.  When it is disabled, the Ps line does not work at all (Without this, the start of ps is bad, alas, please correct it)

PE5PVB, 2024-05-26:
Yes, I found the problem with fastPS off, I'll update the current RC on Discord with a quickfix.

Yurij20, 2024-05-26:
> Yes, I found the problem with fastPS off, I'll update the current RC on Discord with a quickfix.

Thank you) tomorrow I will test and compare more carefully.  Don’t close the topic yet) If something goes wrong again, I’ll make an mpx recording of another station with normal packages) at first glance it really became much better!) Thank you)

Yurij20, 2024-05-27:
> Thank you) tomorrow I will test and compare more carefully. Don’t close the topic yet) If something goes wrong again, I’ll make an mpx recording of another station with normal packages) at first [...]

I tested yesterday's changes and the result was very good, much better, but still not perfect.  As I said earlier, where the very first firmware 1.15 on weakened RDS reception stops the rotation of the dynamic PS, it still often allows mixing of Ps packets.  I made two more recordings of other stations in conditions of heavy RDS reception.  I propose to analyze the situation based on them, perhaps the RDS encoder works correctly in them, unlike yesterday’s entry)

https://disk.yandex.ru/d/TOWtFoZniPBw1A
https://disk.yandex.ru/d/Zz06ai_oJ_ZT8Q

PE5PVB, 2024-05-27:
What I can do is change the function of fast PS. So when enabled always use fastps, when disabled, use it only on first tune. Maybe this will help you?

Yurij20, 2024-05-27:
> What I can do is change the function of fast PS. So when enabled always use fastps, when disabled, use it only on first tune. Maybe this will help you?

Let's do it like this) I once talked about this idea)

PE5PVB, 2024-05-27:
Can you compile yourself?

Yurij20, 2024-05-27:
> Can you compile yourself?

alas, no) if possible, update when it’s convenient)

Yurij20, 2024-05-28:
Tested it.  Everything works as it should now.   Using the new mode of disabled Fast ps I didn’t catch a single error) Thank you for solving this problem :) Now I think the issue is settled.  Thanks again:)

closed 2024-05-28 by PE5PVB