PE5PVB/TEF6686_ESP32 #533: Screensaver in Advanced RDS view does not work

andimik, 2024-04-15:
Seems the screensaver only works in the main screen. But when I am in advanced RDS view, it does not work.

PE5PVB, 2024-04-18:
@ohmytime Can you take a look at this?

ohmytime, 2024-04-26:
After some digging, I found that the original design was to disable the sleep function in the RDS page. 
So, what kind of function is needed now?

[image]

[image]

PE5PVB, 2024-04-29:
@ohmytime Please enable screensaver in all screens except menu.

ohmytime, 2024-05-03:
OK

PE5PVB, 2024-05-07:
Fixed.

PE5PVB, 2024-05-18:
@andimik Can you confirm, so we can close this one?

andimik, 2024-05-18:
Yes, it works now in all menus.

closed 2024-05-18 by andimik