PE5PVB/TEF6686_ESP32 #544: ECC not decoded on UK stations

mrwish7, 2024-04-23:
There seems to be a decode issue with the ECC in the advanced RDS view on UK stations that carry it.

BBC stations transmit the correct ECC flag for the UK (E1), but the TEF always shows ECC: N/A when tuned to them. The fm-dx-webserver (using librdsparser) and RDS Spy both parse the ECC correctly on these transmissions.

[image]
[image]
[image]

andimik, 2024-04-23:
Confirmed. This has already been discussed in https://discord.com/channels/1053804249651359765/1053804431721898056/1230804054452736012 and following posts.

the ECC for UK is listed in `TEF6686.cpp`

```
if (rds.ECC == 225) rds.ECCtext = ECCtext[46];
```

(225 dec is 0xE1)

```
"U.K.", // 46
```

And I've tested an MPX from Croatia (also PI-Code starting with Cxxx, but ECC is E3), there I can see it (Hint: Please ignore Italian in LIC, this is not an error on TEF side).

[image]

And I could not find a problem in the table below. UK is listed only once and E1 with Cxxx is uniq.

Code | case | ECC | ECC_hex | ECC_text | Lookup
-- | -- | -- | -- | -- | --
if (rds.ECC == 162)   rds.ECCtext = ECCtext[129]; | 1 | 162 | A2 | 129 | Anguilla
if (rds.ECC == 240)   rds.ECCtext = ECCtext[179]; | 1 | 240 | F0 | 179 | Aus. ACT
if (rds.ECC == 163)   rds.ECCtext = ECCtext[136]; | 1 | 163 | A3 | 136 | Bolivia
if (rds.ECC == 208)   rds.ECCtext = ECCtext[104]; | 1 | 208 | D0 | 104 | Cameroon
if (rds.ECC == 224)   rds.ECCtext = ECCtext[0]; | 1 | 224 | E0 | 0 | Germany
if (rds.ECC == 225)   rds.ECCtext = ECCtext[1]; | 1 | 225 | E1 | 1 | Greece
if (rds.ECC == 164)   rds.ECCtext = ECCtext[152]; | 1 | 164 | A4 | 152 | Guatemala
if (rds.ECC == 241)   rds.ECCtext = ECCtext[197]; | 1 | 241 | F1 | 197 | Kiribati
if (rds.ECC == 242)   rds.ECCtext = ECCtext[124]; | 1 | 242 | F2 | 124 | Kuwait
if (rds.ECC == 243)   rds.ECCtext = ECCtext[200]; | 1 | 243 | F3 | 200 | Laos
if (rds.ECC == 228)   rds.ECCtext = ECCtext[3]; | 1 | 228 | E4 | 3 | Moldova
if (rds.ECC == 227)   rds.ECCtext = ECCtext[59]; | 1 | 227 | E3 | 59 | Monteneg.
if (rds.ECC == 226)   rds.ECCtext = ECCtext[2]; | 1 | 226 | E2 | 2 | Morocco
if (rds.ECC == 209)   rds.ECCtext = ECCtext[73]; | 1 | 209 | D1 | 73 | Namibia
if (rds.ECC == 210)   rds.ECCtext = ECCtext[83]; | 1 | 210 | D2 | 83 | S. Leone
if (rds.ECC == 160)   rds.ECCtext = ECCtext[226]; | 1 | 160 | A0 | 226 | USA/VI/ PR
if (rds.ECC == 224)   rds.ECCtext = ECCtext[4]; | 2 | 224 | E0 | 4 | Algeria
if (rds.ECC == 162)   rds.ECCtext = ECCtext[130]; | 2 | 162 | A2 | 130 | Antigua.
if (rds.ECC == 240)   rds.ECCtext = ECCtext[180]; | 2 | 240 | F0 | 180 | Aus. NSW
if (rds.ECC == 241)   rds.ECCtext = ECCtext[187]; | 2 | 241 | F1 | 187 | Bhutan
if (rds.ECC == 208)   rds.ECCtext = ECCtext[106]; | 2 | 208 | D0 | 106 | Centr. Af.
if (rds.ECC == 163)   rds.ECCtext = ECCtext[141]; | 2 | 163 | A3 | 141 | Colombia
if (rds.ECC == 225)   rds.ECCtext = ECCtext[5]; | 2 | 225 | E1 | 5 | Cyprus
if (rds.ECC == 226)   rds.ECCtext = ECCtext[6]; | 2 | 226 | E2 | 6 | Czech R.
if (rds.ECC == 228)   rds.ECCtext = ECCtext[8]; | 2 | 228 | E4 | 8 | Estonia
if (rds.ECC == 164)   rds.ECCtext = ECCtext[155]; | 2 | 164 | A4 | 155 | Honduras
if (rds.ECC == 227)   rds.ECCtext = ECCtext[7]; | 2 | 227 | E3 | 7 | Ireland
if (rds.ECC == 209)   rds.ECCtext = ECCtext[126]; | 2 | 209 | D1 | 126 | Liberia
if (rds.ECC == 242)   rds.ECCtext = ECCtext[77]; | 2 | 242 | F2 | 77 | Qatar
if (rds.ECC == 243)   rds.ECCtext = ECCtext[218]; | 2 | 243 | F3 | 218 | Thailand
if (rds.ECC == 160)   rds.ECCtext = ECCtext[226]; | 2 | 160 | A0 | 226 | USA/VI/ PR
if (rds.ECC == 210)   rds.ECCtext = ECCtext[95]; | 2 | 210 | D2 | 95 | Zimbabwe
if (rds.ECC == 224)   rds.ECCtext = ECCtext[9]; | 3 | 224 | E0 | 9 | Andorra
if (rds.ECC == 164)   rds.ECCtext = ECCtext[132]; | 3 | 164 | A4 | 132 | Aruba
if (rds.ECC == 240)   rds.ECCtext = ECCtext[181]; | 3 | 240 | F0 | 181 | Aus. VIC
if (rds.ECC == 241)   rds.ECCtext = ECCtext[227]; | 3 | 241 | F1 | 227 | Bangladesh
if (rds.ECC == 242)   rds.ECCtext = ECCtext[189]; | 3 | 242 | F2 | 189 | Cambodia
if (rds.ECC == 208)   rds.ECCtext = ECCtext[112]; | 3 | 208 | D0 | 112 | Djibouti
if (rds.ECC == 162)   rds.ECCtext = ECCtext[224]; | 3 | 162 | A2 | 224 | EC or BR
if (rds.ECC == 209)   rds.ECCtext = ECCtext[119]; | 3 | 209 | D1 | 119 | Ghana
if (rds.ECC == 163)   rds.ECCtext = ECCtext[156]; | 3 | 163 | A3 | 156 | Jamaica
if (rds.ECC == 229)   rds.ECCtext = ECCtext[63]; | 3 | 229 | E5 | 63 | Kyrgystan
if (rds.ECC == 228)   rds.ECCtext = ECCtext[16]; | 3 | 228 | E4 | 16 | Macedonia
if (rds.ECC == 210)   rds.ECCtext = ECCtext[72]; | 3 | 210 | D2 | 72 | Mozambiq.
if (rds.ECC == 226)   rds.ECCtext = ECCtext[11]; | 3 | 226 | E2 | 11 | Poland
if (rds.ECC == 225)   rds.ECCtext = ECCtext[10]; | 3 | 225 | E1 | 10 | San Marino
if (rds.ECC == 243)   rds.ECCtext = ECCtext[219]; | 3 | 243 | F3 | 219 | Tonga
if (rds.ECC == 227)   rds.ECCtext = ECCtext[12]; | 3 | 227 | E3 | 12 | Turkey
[... 170 lines cut ...]
if (rds.ECC == 226)   rds.ECCtext = ECCtext[52]; | E | 226 | E2 | 52 | Spain
if (rds.ECC == 227)   rds.ECCtext = ECCtext[53]; | E | 227 | E3 | 53 | Sweden
if (rds.ECC == 228)   rds.ECCtext = ECCtext[64]; | E | 228 | E4 | 64 | Turkmenis.
if (rds.ECC == 163)   rds.ECCtext = ECCtext[172]; | E | 163 | A3 | 172 | Turks & C.
if (rds.ECC == 160)   rds.ECCtext = ECCtext[226]; | E | 160 | A0 | 226 | USA/VI/ PR
if (rds.ECC == 164)   rds.ECCtext = ECCtext[175]; | E | 164 | A4 | 175 | Venezuela
if (rds.ECC == 210)   rds.ECCtext = ECCtext[94]; | E | 210 | D2 | 94 | Zambia
if (rds.ECC == 162)   rds.ECCtext = ECCtext[222]; | F | 162 | A2 | 222 | Bahamas
if (rds.ECC == 227)   rds.ECCtext = ECCtext[57]; | F | 227 | E3 | 57 | Belarus
if (rds.ECC == 228)   rds.ECCtext = ECCtext[58]; | F | 228 | E4 | 58 | Bosnia H.
if (rds.ECC == 224)   rds.ECCtext = ECCtext[54]; | F | 224 | E0 | 54 | Egypt
[... cut ...]
if (rds.ECC == 165)   rds.ECCtext = ECCtext[158]; | F | 165 | A5 | 158 | Mexico
if (rds.ECC == 243)   rds.ECCtext = ECCtext[71]; | F | 243 | F3 | 71 | Mongolia
if (rds.ECC == 209)   rds.ECCtext = ECCtext[75]; | F | 209 | D1 | 75 | Nigeria
if (rds.ECC == 226)   rds.ECCtext = ECCtext[56]; | F | 226 | E2 | 56 | Norway
if (rds.ECC == 166)   rds.ECCtext = ECCtext[168]; | F | 166 | A6 | 168 | St. P.& M.
if (rds.ECC == 242)   rds.ECCtext = ECCtext[220]; | F | 242 | F2 | 220 | Vanuatu
if (rds.ECC == 164)   rds.ECCtext = ECCtext[176]; | F | 164 | A4 | 176 | Virgin UK

andimik, 2024-04-23:
The RDS group 1A says,

```
C203 11C0 80E1 9C40
```

`redsea` has no problem, btw.

```
$ cat /media/andreas/Data/temp/bbcr3_mpx.wav | redsea -r 192k | grep ountr
{"pi":"0xC203","group":"1A","country":"gb","has_linkage":true,"prog_item_number":40000,"prog_item_started":{"day":19,"time":"17:00"},"prog_type":"Serious classical","tp":false}
{"pi":"0xC203","group":"1A","country":"gb","has_linkage":true,"prog_item_number":40000,"prog_item_started":{"day":19,"time":"17:00"},"prog_type":"Serious classical","tp":false}
 ```

andimik, 2024-04-23:
It has nothing to do with the dot(s), because I have re-flashed my TEF using "United K" instead of "U.K."

PE5PVB, 2024-05-02:
Fixed.

andimik, 2024-05-03:
Has been confirmed on discord, issue can be closed.

https://discord.com/channels/1053804249651359765/1053804431721898056/1235670401674121246

[image]

Explanation of the bug:

> there is a counter which tells the radio if it is about to receive ECC, LIC or other values. These are in bit 12, 13 and 14. But I also parsed bit 15 by mistake. As long as bit 15 is 0 all is ok, but on BBC bit 15 is set to 1 and then the parsing will fail. So I had to ignore bit 15.

closed 2024-05-03 by PE5PVB