PE5PVB/TEF6686_ESP32 #556: Auto squelch is activated by default on silver TEFs with no squelch knob, cannot be deactivated. AM tuning issue.

MCelliotG, 2024-05-09:
Two issues I encountered with the latest beta.
1. The silver TEFs that have no squelch knob now have squelch activated by default and the option to deactivate is not available. Perhaps it would be better to set default squelch value to off, to support these models as well.
2. The tuning cycle on AM band is broken. When the tuning reaches the end of the band (1791 khz) and returns to 522 khz it blocks and the tuning knob cannot tune the frequencies upwards, only downwards. The only way to reactivate the knob tuning upwards is to manually select the frequency, but when the tuning reaches the end of the band it blocks again. I could make a video with the issue.

PE5PVB, 2024-05-10:
1.  Please do a factory reset.
2.  Fixed.

MCelliotG, 2024-05-10:
Factory reset fixes the first issue but it defaults the menus to Chinese. This should be considered as well.

PE5PVB, 2024-05-10:
Check the file config.h if the line #define CHINA_PORTABLE is remarked.

ohmytime, 2024-05-10:
> Factory reset fixes the first issue but it defaults the menus to Chinese. This should be considered as well.

I'm considering reverting to the previously selected language when doing a factory reset.

PE5PVB, 2024-05-10:
> > Factory reset fixes the first issue but it defaults the menus to Chinese. This should be considered as well.
> 
> I'm considering reverting to the previously selected language when doing a factory reset.

Well, default language after a reset should just be English. Default settings call should set all eeprom values to a designated value. No recovery.

ohmytime, 2024-05-10:
> Well, default language after a reset should just be English. Default settings call should set all eeprom values to a designated value. No recovery.

Well we can make it the second parameter of function DefaultSettings, 
```
DefaultSettings(byte userhardwaremodel, byte nowlanguage)

```
So that we can reset user pre-selected language.

ohmytime, 2024-05-10:
Think about this case:
User selected Italian as his display language, and then he made a factory reset.
If the default language is English ,it is okay, but I think italian would be better.

andimik, 2024-05-10:
Better than Chinese for sure.

PE5PVB, 2024-05-10:
> Think about this case: User selected Italian as his display language, and then he made a factory reset. If the default language is English ,it is okay, but I think italian would be better.

And what if a user set a foreign language by accident and has no way how to find the language setting?

ohmytime, 2024-05-10:
> > Think about this case: User selected Italian as his display language, and then he made a factory reset. If the default language is English ,it is okay, but I think italian would be better.
> 
> And what if a user set a foreign language by accident and has no way how to find the language setting?

BTW, do we now 

> > Think about this case: User selected Italian as his display language, and then he made a factory reset. If the default language is English ,it is okay, but I think italian would be better.
> 
> And what if a user set a foreign language by accident and has no way how to find the language setting?

The need does exist.
BTW, do we need to, for example, make a detailed breakdown of restoring factory settings?

ALL(default to English)
User Settings + Preset stations
User Settings only 

It's probably better to do it on the menu.