PE5PVB/TEF6686_ESP32 #569: Text ticker not optimized

RZCH-DX, 2024-05-19:
Hello,

The ticker feature is not very optimized. I have noticed some issues:

- If there is only one AF freq, it will appear two times in the advanced RDS view. 
- Radiotext repeats if it's terminated with 0x0D
- Ticker will sometimes scroll even if the text isn't overflowing the space
- Sometimes some character(s) is/are appended a few spaces after the RT. Retuning gets rid of it.

These issues seems to happen with all tickers in the firmware.
Tested with 2.10 RC3

All issues except the unnecessary scrolling were not present in version 2.00

Examples:
[image]

[image]
[image]

RT not scrolling but with trailing hyphen that should not be there:
https://youtu.be/cDeJrWqslc0

RT scrolling even if there's enough space for it without scrolling:
https://youtu.be/bzz1613KdZc

PE5PVB, 2024-05-19:
1.  This is caused by continiously scrolling. I will change this when only one AF is present.
2.  This is as expected. 
3.  It's difficult to determ the exact textlength, because fontwidth is dynamic. I had to set a rule for this.
4.  I think the radiostation is also transmitting RT method B without content.

RZCH-DX, 2024-05-19:
Hello Sjef,

Thank you for the quick reply. 

Regarding issue 2: It's confusing to see the same text two times. The start and end of the received RT is not clearly visible anymore. 
Maybe it's an option just to scroll until the last character is reached. Then restarting the ticker.

Issue 4: Do you mean RT in group 2B? The station is only transmitting 0A, 2A and 4A groups. No 2B.
It alternates between RT flag A and B, though with the same content as seen in the RDS spy screenshot.

PE5PVB, 2024-05-19:
Can you send me a MPX file of that channel?

RZCH-DX, 2024-05-19:
Yes, of course
[4f29190524.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/15369506/4f29190524.zip)

PE5PVB, 2024-05-19:
[image]

Seems they broadcast a RT line starting with - sometimes. What other problems are hrere?

RZCH-DX, 2024-05-19:
Ah I didn't mean the hyphen at the beginning. The TEF handles this one without any problems.

I'm talking about this one, which is not transmitted and not seen in RDS spy:
[image]

PE5PVB, 2024-05-19:
OK. I watched the whole MPX sample you send me, but was not able to reproduce it.

RZCH-DX, 2024-05-20:
I've found a way to reproduce it. 

To reproduce it, you need a station which sends out 0x0D terminated RT, alternates between RT flag A/B and doesn't send anything (nothing, not even space letters) after 0x0D. (the majority of the NL stations seem to do it wrong)
You can use the MPX recording from before. It checks all the boxes.

1. Set "Show RDS errors" to "All" on the TEF
2. Null out the signal to the point where RDS is weak and spits out garbage.
3. Stay there for some seconds. 
4. Then move the antenna somewhere where the RDS signal is strong enough for errorless data. 
5. When the station sends a new RT flag, there will be random characters stuck from before. These characters are always after 0x0D.

This should not happen, a new RT flag should wipe out all previous data completely.

PE5PVB, 2024-05-20:
Well, if RT buffer is enabled you will always see the previous RT. So maybe that is what you have experienced. But you are right, currently the RT is not taking 0x0D as an end character, I'll fix that.

RZCH-DX, 2024-05-20:
RT buffer was disabled the entire time.

RZCH-DX, 2024-05-22:
With RC4, the DoubleAF-bug is fixed. But it's still present in the EON ticker.

The radiotext in the ticker now repeats even sooner. So this got worse.
[image]
[image]

PE5PVB, 2024-05-22:
I guess there is only one EON in this stream? Then it's normal to see this, because ticker is endless.

RZCH-DX, 2024-05-25:
The unnecessary scrolling issue has been fixed with RC6.

The issue with the trailing characters which aren't going away with RT flag refresh is still there though.

PE5PVB, 2024-05-25:
> The issue with the trailing characters which aren't going away with RT flag refresh is still there though.

I need a MPX file to reproduce and fix that.

lxsxl, 2024-05-25:
The ODA-ticker in EON mode is very slow: it moves two or three characters, than stops for a moment, then the next few characters and so on.

RZCH-DX, 2024-05-25:
> > The issue with the trailing characters which aren't going away with RT flag refresh is still there though.
> 
> I need a MPX file to reproduce and fix that.

Use this MPX file and the following instruction to reproduce it
[4f382505242223.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/15444935/4f382505242223.zip)

> 1. Set "Show RDS errors" to "All" on the TEF
> 2. Null out the signal to the point where RDS is weak and spits out garbage.
> 3. Stay there for some seconds.
> 4. Then move the antenna somewhere where the [...]

PE5PVB, 2024-05-26:
OK. I added a detector which should delete all characters after the first find of 0x0d in the Radiotext.

andimik, 2024-05-26:
With RC7 from discord the ticker exceeds the "cell" in normal view and overlaps the flag dots. In advanced RDS view it's OK.

PE5PVB, 2024-05-26:
Yes I know. Already fixed.

lxsxl, 2024-05-26:
The continously scrolling texts are really great! I didn't see any problems with the last code.

RZCH-DX, 2024-05-26:
> OK. I added a detector which should delete all characters after the first find of 0x0d in the Radiotext.

It's still happening with the latest commit

lxsxl, 2024-05-26:
Does it really? Or is it only the same text that is displayed in a loop? This is intended!

PE5PVB, 2024-05-26:
Are you  able to monitor via Serial, like with Putty? in that case I can make a version for you with some debug output.

RZCH-DX, 2024-05-26:
> Does it really? Or is it only the same text that is displayed in a loop? This is intended!

No, it's not the same text. It's garbage data which is stuck in the RT ticker until retuning. 
Reproduce it with the instructions 7 comments above.

This also happens with stations without 0x0D. 
These garbage characters sometimes appear after the 64th character.

The station in the following video doesn't use 0x0D and always send out 64 characters in RT. I could produce some garbage with the method above, which cause the RT in the ticker to be longer than 64 characters. (Just count the dashes after "WORLD")
After retuning, garbage is gone and RT is 64 characters again.

https://youtu.be/TZc85ix_Nr8

Maybe this issue can be fixed by limiting the radiotext string to 64 characters.

PE5PVB, 2024-05-26:
Please check if you see the dot on 2B turn on. I think because of the worse signal in combination with show all errors, the receiver thinks there is a Radiotext 32 chars in group 2B and fills it with garbage. When 32 chars RT is detected, it is merged to the 64 character RT from group 2A.

RZCH-DX, 2024-05-26:
> Please check if you see the dot on 2B turn on. I think because of the worse signal in combination with show all errors, the receiver thinks there is a Radiotext 32 chars in group 2B and fills it with garbage. When 32 chars RT is detected, it is merged to the 64 character RT from group 2A.

Yes, 2B appears.

PE5PVB, 2024-05-26:
There is the problem. Note that when you set Show all errors, things like this can happen. What happens is that the wrong RDS group is processed. Can't fix that when the data is corrupt.

RZCH-DX, 2024-05-26:
Yes of course. But why is 2B-radiotext appended to 2A-radiotext?

PE5PVB, 2024-05-26:
Yes it is merged with 2A, but if the software detects a character at for example, position 20, then you have this result. That's why there is error correction on RDS. But if you switch it off........ ;-)

RZCH-DX, 2024-05-26:
> Yes it is merged with 2A, but if the software detects a character at for example, position 20, then you have this result. That's why there is error correction on RDS. But if you switch it off........ ;-)

Currently the entire 2B RT is just appended after the 64 characters of 2A RT.
Let's say the false 2B character at position 20 is Ä

Currently looks like this:

Dua_Lipa_-_Houdini_(2023)_______________________________________ ___________________Ä____________

instead of being merged into like here:

Dua_Lipa_-_Houdini_Ä2023)_______________________________________

PE5PVB, 2024-05-26:
You don't want to merge it like that. There are stations who sends different texts on 2A and 2B.

RZCH-DX, 2024-05-26:
> You don't want to merge it like that. There are stations who sends different texts on 2A and 2B.

All right, understood. Then everything is resolved for me. Thank you very much!

closed 2024-05-26 by RZCH-DX