PE5PVB/TEF6686_ESP32 #579: Issues in ODA ticker

lxsxl, 2024-05-26:
There are two issues in ODA ticker:
1. ticker is moving unsteady each time there are updates in AF-list or in EON-list.
2. there is a grid line that likely is not intended:
[image]

lxsxl, 2024-05-26:
Again to the first point: in EON mode the ticker is unsteady in any case even if there is no RDS reception. Ticker moves for two or three characters than pauses for a moment (and so on). In AF mode it pauses only if new AF are inserted.

PE5PVB, 2024-05-26:
I'll fix the canvas. Problem with the stuttering ticket is not easy to solve, because the software has to check the eon table for changes continiously. Currently it's done every 1 second. I changed it to 2 seconds.

lxsxl, 2024-05-26:
Just uploaded RC8. Everything looks fine now! I close this issue

closed 2024-05-26 by lxsxl