PE5PVB/TEF6686_ESP32 #582: RDS quality indicator on main screen

SpiderBlocker, 2024-05-28:
Not really a big issue, but I have been wondering why the green RDS quality indicator bars on the main screen must have [url=https://imgur.com/a/s2GnkUY]these small holes[/url] to them?

PE5PVB, 2024-05-28:
Checked it. But maybe it's a display thing. I can't reproduce it on my radio. Nothing to do about it. It looks kinda cool ;-)

PE5PVB, 2024-05-28:
Fixed

SpiderBlocker, 2024-05-28:
Wunder**bar**!

lxsxl, 2024-05-28:
Almost the same "issue" is in advanced RDS mode (here only the top ends of the bars).

SpiderBlocker, 2024-05-30:
> Almost the same "issue" is in advanced RDS mode (here only the top ends of the bars).

Indeed, it's there as well. Maybe it is some sort of relic ;-)

PE5PVB, 2024-05-30:
Did you check out the update from yesterday?

SpiderBlocker, 2024-05-30:
> Did you check out the update from yesterday?

Yes, it was fixed on the main screen, but not on the advanced RDS screen.

SpiderBlocker, 2024-06-01:
> Did you check out the update from yesterday?

Main screen was fixed, but its still there on the advanced RDS screen:
[image]

PE5PVB, 2024-06-01:
Fixed.

lxsxl, 2024-06-02:
There is still a little difference in the bars between standard and advanced mode:
[image]
[image]

SpiderBlocker, 2024-06-03:
RC13, bummerrrszz! Holes are back; both on main screen and advanced RDS. Someone is poking his little drill into them every time!
[image]

PE5PVB, 2024-06-04:
I redesigned this section so only one full bar with rounded corners is written to the screen for each error level step. This will fix this problem once and for all. If it still happens, than your display is faulty ;-)

SpiderBlocker, 2024-06-04:
Hatseflats, you fixed it again! 
As I am assuming with you that it now has been fixed once & for all, let's close this topic accordingly :-)

closed 2024-06-04 by SpiderBlocker