PE5PVB/TEF6686_ESP32 #584: The performance of the encoder has deteriorated when rebuilding and switching saved channels.

Yurij20, 2024-05-29:
Maybe I just noticed it, but with the generally slow restructuring and switching of saved channels, encoder clicks are often ignored and in any situation the first click is consistently missed with each change in the direction of rotation of the encoder.

andimik, 2024-05-29:
Seems this is related to hardware. Please clean the Encoder with a specific fluid. I had the same problem.

Yurij20, 2024-05-29:
> Seems this is related to hardware. Please clean the Encoder with a specific fluid. I had the same problem.

I recently used firmware 1.18, one of the old ones for comparison, in it the encoder works fine on my device

andimik, 2024-05-29:
Have you tried to use a different model in the settings?

Yurij20, 2024-05-29:
> Have you tried to use a different model in the settings?

Yes, sure.  I read somewhere on the forums that the problem with skipping occurs due to the autosave mode working in a split second.  If you change it to at least 3 seconds, then the encoder will work much better when adjusted.  Let me remind you once again that the encoder scrolls through the menu perfectly

PE5PVB, 2024-05-29:
Well, the radio has much to do when switching between memory positions. When you are turning too fast this can end up in not responding. What if you tune preset by preset?

Yurij20, 2024-05-29:
> Well, the radio has much to do when switching between memory positions. When you are turning too fast this can end up in not responding. What if you tune preset by preset?

Here the situation is such that encoder clicks are ignored not only when switching presets, but during normal settings and mainly when I rotate the encoder quietly.  With fast rotation, relatively everything is better

PE5PVB, 2024-05-29:
Are you sure you didn't activate optical encoder?

Yurij20, 2024-05-29:
> Are you sure you didn't activate optical encoder?

switching this optical/standard mode does not make any difference

PE5PVB, 2024-05-29:
Since v2.00 I've added some extra protection for wear and old rotary encoders, to prevent glitches. Maybe you run into this.

Yurij20, 2024-05-30:
> Since v2.00 I've added some extra protection for wear and old rotary encoders, to prevent glitches. Maybe you run into this.

Maybe, but here on a forum one friend modified firmware 1.15 and I tested it and the situation was exactly the same with the encoder, I also explained to him about the problem, as he explained to me, this happens due to the minimum autosave threshold after rebuilding the receiver, he raised it to  3 seconds and the problem was solved, the encoder began to work perfectly) Maybe I can try this and maybe it will help you?) If not, I’ll let you know and return everything back and leave everything as is

PE5PVB, 2024-06-01:
I tried to win some time by not refreshing the whole page when swithcing bands. Maybe this will help you a little. For the rest, I see nothing more to improve right now. If someone have better ideas, please let me know.

Yurij20, 2024-06-01:
> I tried to win some time by not refreshing the whole page when swithcing bands. Maybe this will help you a little. For the rest, I see nothing more to improve right now. If someone have better ideas, please let me know.

Okay, right now I don’t have my computer handy on a trip, I’ll test the latest update tomorrow.  I remind you that the problem is acute with the encoder adjustment on the very first version of the Chinese receivers.  The menu rotates the encoder perfectly.  Below is a photo of my receiver that I use in the exact series where this problem occurs.  On other later devices everything is relatively good
[image]

PE5PVB, 2024-06-02:
Maybe in the design with problems they didn't add caps on the rotary encoder to reduce bouncing.

Yurij20, 2024-06-02:
> Maybe in the design with problems they didn't add caps on the rotary encoder to reduce bouncing.

I checked it now, it’s better with the encoder, it’s faster to change tuning, but there are still clicks with slow tuning.  Maybe you can still try changing autosave after restructuring to 3 seconds?)

PE5PVB, 2024-06-02:
No, that's not causing any delay, only a few uS.

Yurij20, 2024-06-02:
> No, that's not causing any delay, only a few uS.

ok, then we'll leave it like that.  I see you made adjustments for optical encoders, it helped)

closed 2024-06-03 by Yurij20