PE5PVB/TEF6686_ESP32 #588: When returning from extended RDS and from the menu to the main screen, a slight stutter appeared and the RDS block began to reset

Yurij20, 2024-06-02:
When returning from extended RDS and from the menu to the main screen, a slight stutter appeared and the RDS block began to reset.

PE5PVB, 2024-06-02:
Yes, stutter is known. Can't change that.

Yurij20, 2024-06-02:
> Yes, stutter is known. Can't change that.

before rc12 there was no such problem and the RDS block was not reset in this situation

PE5PVB, 2024-06-02:
Yup. This is the effect of not refreshing the screen on memory change. Sorry.

Yurij20, 2024-06-02:
> Yup. This is the effect of not refreshing the screen on memory change. Sorry.

I suggest undoing the changes that caused this problem(

SpiderBlocker, 2024-06-02:
> > Yup. This is the effect of not refreshing the screen on memory change. Sorry.
> 
> I suggest undoing the changes that caused this problem(

I hope a fix would be possible somehow indeed. That stutter has a kind of sloppy feel to it and it wasn't there in the first place.

Yurij20, 2024-06-03:
> > > Yup. This is the effect of not refreshing the screen on memory change. Sorry.
> > 
> > 
> > I suggest undoing the changes that caused this problem(
> 
> I hope a fix would be possible somehow indeed. That stutter has a kind of sloppy feel to it and it wasn't there in the first place.

Okay, you can put up with the stuttering, at least so that the RDS unit does not reset when returning from the extended RDS

PE5PVB, 2024-06-03:
Made a workaround for you.
https://github.com/PE5PVB/TEF6686_ESP32/commit/83f2888887de895e52a0f2bdcb02be29ed2606f4

Yurij20, 2024-06-03:
fixed) thanks

closed 2024-06-03 by Yurij20