PE5PVB/TEF6686_ESP32 #589: Please make changes to the translation of some PTY positions into Russian:

Yurij20, 2024-06-02:
Please make changes to the translation of some PTY positions into Russian:

Varied Speech – Разное
Pop Music – Поп музыка
Rock Music – Рок Музыка
Light Classic M – Легкая классика
Serious Classical – Классика
Other Music – Другая музыка
Children’s Programs – Детские программы
National Music – Национальная музыка
Oldies Music – Музыка ретро

This way these positions are translated more accurately :)

andimik, 2024-06-02:
What was the source of the existing texts?

PE5PVB, 2024-06-02:
> What was the source of the existing texts?

Official RDS documentation. But others also found errors in that doc.

PE5PVB, 2024-06-03:
Done. See: https://github.com/PE5PVB/TEF6686_ESP32/commit/b0332a4de735e128f6f7f5dd47e6cfa2d58f062f

Yurij20, 2024-06-03:
fixed) thanks

closed 2024-06-03 by Yurij20