PE5PVB/TEF6686_ESP32 #591: AutoScan plug in cannot restart scanning once a signal if found

numerci, 2024-06-04:
Preconditions:
A. DX scan options: Memory channels=Only
B. DX scan options: Auto cancel scan = Signal or Correct PI
Steps to reproduce:
1. Start an AutoScan using the WebServer plug in.
2. Wait till a signal is found (constant signal).
3. Press again the AutoScan button from the WebServer.
Result: The scan doesn't start anymore.
Note: Probably this appeared a few versions back when the fix to start from the last user frequency was added. Actually it should be the next one otherwise we enter a loop.

numerci, 2024-06-04:
It does the same on TEF after a DX Scan. if it stops and if you press the Band button normally it starts again but in this case is not starting like the AutoScanner does.

PE5PVB, 2024-06-04:
Fixed.

numerci, 2024-06-04:
Thank you very much :)

numerci, 2024-06-05:
Works fine both from the TEF and from the AutoScanner on RC14. Closed

closed 2024-06-05 by numerci